Opentrons/opentrons #10682: 6.0 Feedback: inconsistent font size in protocol description

ecormany, 2022-06-08:
### Overview

The font size is different in the protocol detail description depending on whether the "Read More" or "Read Less" view is toggled.

### Current Behavior

Clicking the "Read Less" or "Read More" text button causes the font size of the description to change. When the description is short, this means the only change when clicking the button is the font size change.

https://user-images.githubusercontent.com/1958332/172695948-77df3675-0ee9-45c6-9d21-baa15843d47d.mov

 

### Expected Behavior

The font size should either:
- never change
- decrease in size when clicking "Show More" and there is overflow text (opposite of current behavior!)

Additionally the "Show More" button could be disabled or invisible if there is no overflow text.

### Steps To Reproduce

1. Click on a protocol in the Protocols list
2. Click the "Read More" and "Read Less" text buttons on the protocol detail page

### Operating system

Mac

aptrons22, 2022-06-16:
Fixed by https://github.com/Opentrons/opentrons/pull/10805

closed 2022-06-16 by aptrons22

nusrat813, 2022-06-29:
Verified fixed