Opentrons/opentrons #10688: RFC: simplify `SynchronousAdapter` so that it doesn't wrap objects

mcous, 2022-06-08:
## Overview

The [`opentrons.hardware_control.adapters.SynchronousAdapter`](https://github.com/Opentrons/opentrons/blob/212a0f4798e5937cba0a4a2c25be379fd5c25722/api/src/opentrons/hardware_control/adapters.py) class exists for two purposes:

- To allow synchronous interfaces, like the v2 `ProtocolContext`, to call asynchronous interfaces like the hardware control APIs synchronously
- To ensure those synchronous calls are run in the Hardware Control dedicated thread, managed by the [`ThreadManager`](https://github.com/Opentrons/opentrons/blob/212a0f4798e5937cba0a4a2c25be379fd5c25722/api/src/opentrons/hardware_control/thread_manager.py)

It accomplishes this by:

- Wrapping up an object with asynchronous methods
- Intercepting any calls to that object and sending them through i`asyncio.run_coroutine_threadsafe`](https://docs.python.org/3/library/asyncio-task.html#asyncio.run_coroutine_threadsafe)

So, a caller (like the `ProtocolContext`) will be given a `HardwareControlAPI` object wrapped up in a `SynchronousAdapter` rather than an actual `HardwareControlAPI`. This has the following downsides:

- Type checking information is completely erased
    - To some extend, this reduces system safety
    - More importantly (IMO), it severely hampers the DX (developer experience) when working in important objects like the `ProtocolContext`
- If some method of the wrapped object returns another asynchronous class instance, that instance then needs to be wrapped in its own `SynchronousAdapter`
    - For example: the `ProtocolContext` must get attached module objects from the `HardwareControlAPI` and then wrap each of them up in a `SynchronousAdapter`

This RFC proposes and evolution of the `SynchronousAdapter` interface to avoid these serious pitfalls and exchange them for smaller ones.

## Proposal

Create a `SynchronousAdapter` class that takes a loop and gives you an object with a `call_sync` method.

```python
import asyncio
from typing import Any, Callable, Coroutine, TypeVar
from typing_extensions import ParamSpec

AsyncParamsT = ParamSpec("AsyncParamsT")
AsyncReturnT = TypeVar("AsyncReturnT")

class SynchronousAdapter:
    """An adapter to allow asynchronous function to be called synchronously.
    
    Args:
        loop: The event loop to run the asynchronous function in.
    """
    def __init__(self, loop: asyncio.AbstractEventLoop) -> None:
        self._loop = loop
   

    def call_sync(
        self,
        func: Callable[AsyncParamsT, Coroutine[Any, Any, AsyncReturnT]],
        *args: AsyncParamsT.args,
        **kwargs: AsyncParamsT.kwargs,
    ) -> AsyncReturnT:
        return asyncio.run_coroutine_threadsafe(
            func(*args, **kwargs)
            loop=self._loop,
        )
```

From there, add a method to the `ThreadManager` to get a `SyncAdapter` using the managed thread/loop:

```python
class ThreadManager:
    ...
    def create_sync_adapter(self) -> SynchronousAdapter:
        return SynchronousAdapter(loop=self._loop)
```

Here is the most complicated part of the proposal. Where anything is expecting a `SynchronousAdapter[SomethingT]`, change it to depend on a `SynchronousAdapter` _and_ a `SomethingT`. So, taking [`ProtocolContextImplementation`](https://github.com/Opentrons/opentrons/blob/e2114d50d00f420cc67a765e445e9052b779483b/api/src/opentrons/protocols/context/protocol_api/protocol_context.py) as an example:

```python
class ProtocolContextImplementation(AbstractProtocol):
    def __init__(
        self,
        sync_adapter: SynchronousAdapter,
        hardware: HardwareControlAPI,
        api_version: Optional[APIVersion] = None,
        bundled_labware: Dict[str, LabwareDefinition] = None,
        bundled_data: Dict[str, bytes] = None,
        extra_labware: Dict[str, LabwareDefinition] = None,
    ) -> None:
        ...

    def delay(self, seconds: float, msg: Optional[str]) -> None:
        """Delay execution for x seconds."""
        self._sync_adapter.call_sync(self._hardware.delay, seconds=seconds)
```

## Upsides

- Explicit interface rather than implicit interface
- Typing information preserved
- DX stuff like cmd+click to jump to definition and docstrings on tooltips works in editors like VSCode and PyCharm
- The type signatures of module contexts will no longer [be a lie](https://github.com/Opentrons/opentrons/blob/212a0f4798e5937cba0a4a2c25be379fd5c25722/api/src/opentrons/protocol_api/module_contexts.py#L346-L348) that has to be `# type: ignore`'d
- Since `SynchronousAdapter` only does one thing, alternative implementations can be explored and prototyped
    - For example, a cross-process ZMQ-based adapter

## Downsides

- More (keyboard) typing 🙃 
- Increased risk of doing `self._hardware.whatever()` without wrapping it in `self._sync_adapter.call_sync`
    - Mitigation: if there are unit tests covering the method in question, this mistake will raise errors/warnings
    - Mitigation: newer versions of mypy can also warn against some variants of this

sfoster1, 2022-06-09:
This makes a ton of sense and will be a great improvement - so glad `ParamSpec` exists now.

Quite honestly a similar thing should happen with the `ThreadManager`.

mcous, 2022-06-10:
Talked about this with @sfoster1 and we agree that this is a good move. Marking this ticket as a `refactor` ticket and removing the RFC label

mcous, 2022-06-11:
Proposal to make this change in advance of `ParamSpec` support. I think longer term, the proposal in the original ticket is more flexible and readable, but in the mean time, this should work equivalently:

```python
import asyncio
from typing import Any, Coroutine, TypeVar

AsyncReturnT = TypeVar("AsyncReturnT")

class SynchronousAdapter:
    """An adapter to allow asynchronous function to be called synchronously.
    
    Args:
        loop: The event loop to run the asynchronous function in.
    """
    def __init__(self, loop: asyncio.AbstractEventLoop) -> None:
        self._loop = loop
   
    def call_sync(self, coro: Coroutine[Any, Any, AsyncReturnT]) -> AsyncReturnT:
        return asyncio.run_coroutine_threadsafe(coro, loop=self._loop)
```

```python
class ProtocolContextImplementation(AbstractProtocol):
    def __init__(
        self,
        sync_adapter: SynchronousAdapter,
        hardware: HardwareControlAPI,
        api_version: Optional[APIVersion] = None,
        bundled_labware: Dict[str, LabwareDefinition] = None,
        bundled_data: Dict[str, bytes] = None,
        extra_labware: Dict[str, LabwareDefinition] = None,
    ) -> None:
        ...

    def delay(self, seconds: float, msg: Optional[str]) -> None:
        """Delay execution for x seconds."""
        self._sync_adapter.call_sync(self._hardware.delay(seconds))
```

closed 2022-08-03 by TamarZanzouri