Opentrons/opentrons #10714: 6.0 Feedback:  Robot upgrade fails after 20min - blocked unable to perform alpha testing

reachojhaneha, 2022-06-10:
### Overview

I am getting errors when upgrading robot to alpha-2, leading to upgrade failure  with write EPIPE error. It took close to 20min for "Robot Update in progress" modal followed by error.

### Current Behavior

[image]
[image]
[image]
[image]

### Expected Behavior

Robot Update should be successful

### Steps To Reproduce

Steps to reproduce:

1. Upgrade the app
2. App goes for restart
3. Click on device -> go to device landing page
4. Click on “view update ” link from software update banner
5. “Robot update in progress” modal shows up and after 20min fails with write EPIPE error
6. console shows 409 conflict with /server/update/begin endpoint

### Operating system

Mac

laviera, 2022-06-10:
@reachojhaneha - these errors look somewhat network related ...
1.  What version are u upgrading from ?
2. can u post the logs from the robot ?

reachojhaneha, 2022-06-10:
@laviera I am upgrading from alpha-1
Here are the logs from robot
[api.log](https://github.com/Opentrons/opentrons/files/8878744/api.log)
[serial.log](https://github.com/Opentrons/opentrons/files/8878745/serial.log)
[server.log](https://github.com/Opentrons/opentrons/files/8878747/server.log)

reachojhaneha, 2022-06-10:
[image]
I was able to upgrade my robot with USB cable. 
Closing issue

closed 2022-06-10 by reachojhaneha