Opentrons/opentrons #10718: question:

m89ozturk, 2022-06-10:
Hi Everyone,
I am trying to build an ssh connection to the OT-2 using paramiko but I receive the following error.
Any help is greatly appreciated.

CODE:
ssh = paramiko.SSHClient()
ssh.load_host_keys('.ssh/known_hosts')
ssh.set_missing_host_key_policy(paramiko.AutoAddPolicy())
SSHkey = paramiko.RSAKey.from_private_key_file("ot2_ssh_key", password="XYZ")
ssh.connect(hostname='169....', username='root', passphrase='XYZ', pkey=SSHkey)

ERROR:
File ".\client.py", line 435, in connect
self._auth(
File ".\client.py", line 766, in _auth
raise saved_exception
File ".\client.py", line 666, in _auth
self._transport.auth_publickey(username, pkey)
File ".\transport.py", line 1635, in auth_publickey
return self.auth_handler.wait_for_response(my_event)
File ".\auth_handler.py", line 259, in wait_for_response
raise e
paramiko.ssh_exception.AuthenticationException: Authentication failed.

mcous, 2022-06-10:
@y3rsh you were doing some Paramiko stuff recently, did you ever get that working?

pkMyt1, 2022-06-10:
We use this method as well.  I am not a paramiko expert but it looks like there might be something wrong with your key.
We initialize the communication with
`    def connect_to_ot2(self):
        """
        Establish SSH connection to robot.
        :return:
        """
        robot_name = "OT2CEP20180915A20"
        try:
            host_ip = socket.gethostbyname(robot_name)
        except socket.gaierror:
            self.error_report("Unable to connect to Opentrons OT-2 {}\n Is robot on and connected to computer?"
                              .format(robot_name))
            return

        ssh_client = SSHClient()
        ssh_client.load_system_host_keys()
        ssh_client.set_missing_host_key_policy(AutoAddPolicy())
        try:
            ssh_client.connect(hostname=host_ip, username='root',
                               key_filename='C:{0}Users{0}robotron{0}ot2_ssh_key'.format(os.sep))
        except OSError:
            self.error_report("SSH unable to establish connection to robot.  Secure Key error")
            self.critical_error = True
            raise SystemExit(1)

        return ssh_client`

We use the connection to transfer a parameter file to the OT-2 for the run using
`    def transfer_tsv_file(self):
        """
        If everything checks out then transfer the files to the robot.
        :return:
        """

        # Establish a connection to the robot.
        self.ssh_client = self.connect_to_ot2()

        # If communications with the OT-2 cannot be established then let the user know so they can decide.
        if not self.ssh_client:
            self.warning_report("Communications with OT-2 not established.  If this was expected then you can safely "
                                "ignore this message")
            return

        # If we have a critical error then don't do anything else.
        if self.critical_error:
            return

        if not self.path_to_tsv:
            self.select_file()

        # program_path = os.path.dirname(self.path_to_program)
        program_name = os.path.basename(self.path_to_program)

        # Initialize scp and transfer the files to the robot.
        scp = SCPClient(self.ssh_client.get_transport())
        # TSF file transfer
        scp.put(files=self.path_to_tsv, remote_path="{}{}".format(self.server_path, self.server_tsv_file),
                preserve_times=True)

        # Program file transfer
        # scp.put(files=self.path_to_program, remote_path=self.server_path, preserve_times=True)

        # Confirm files have transferred.
        cmd = "ls {}".format(self.server_path)
        stdin, stdout, stderr = self.ssh_client.exec_command(cmd)
        stdout.channel.recv_exit_status()
        response = stdout.readlines()
        transferred_files = []

        for line in response:
            transferred_files.append(line.strip("\n"))

        if self.server_tsv_file not in transferred_files:
            self.error_report("TSV File Transfer Failed")
            self.critical_error = True
            # elif program_name not in transferred_files:
            # self.error_report("Program File {} Transfer Failed".format(program_name))
            # self.critical_error = True
        else:
            self.success_report("All Files Transferred Successfully", "File Transfer")

        scp.close()`
The full code is in our GitHub [https://github.com/Gaorav-Gupta-Lab/Opentrons_GUI]

mcous, 2022-06-10:
@m89ozturk feel free to continue posting / asking questions in this thread! Hopefully the above example and link can help you get your setup working

closed 2022-06-10 by mcous