Opentrons/opentrons #10730: 6.0 Feedback:  [Design QA] Device Details Page#10150

mmencarelli, 2022-06-10:
### Current Behavior

[Annotated Figma frame](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=14185%3A283354)

- [ ] (1) Just want to note that when I click on the 'x', nothing happens
- [ ] (2) Should be 1rem margin here
- [ ] (3) With the addition of the toast, the menu overflow is off and doesn't have the 4px margin on the top and right side 
- [ ] (4) This image should not change placement when there's a toast message present
- [ ] (5) This svg should be 20x20 vs 16x16 
- [ ] (6) Enabled colour in our toggle component should be darkGreyEnabled
- [ ] (7) Update colour of h6 to darkGreyEnabled
- [ ] (8) Remove box-shadow
- [ ] (9) Extra margin on right/left side. Looks like we have 4px on `PipetteCard_P10 Single-Channel GEN1` and 8px on `Flex-sc-1qhp8l7-0 lfWeqr`
- [ ] (10) update `Flex-sc-1qhp8l7-0 kEIVxu` colour of border to our mediumGreyEnabled
- [ ] (11) This open menu needs to be placed 4px below and aligned with the button. Check out the documentation in the design system for placement 
- [ ] (12) There's a bit of space to the left of the toggle that makes it indent to the right from the header above it
- [ ] (13) Could we update the image for the OT-2 to a retina image? It's looking blurry on my screen. I'll attach a new image with the same ratio at double the size. 
- [ ] (14) Wanted to flag this  (664px width)
- [ ] (15) Wanted to flag this (952px width) 

[image]
[image]
[image]
[image]

vabruzzo, 2022-06-16:
this issue can also address #10771

vabruzzo, 2022-06-22:
1 will be fixed when #10732 is merged in.

commit a37c5c26 (2022-06-24): style(app): address device details page design QA (#10842)

Closes #10730, #10771
closed 2022-06-24 by vabruzzo