Opentrons/opentrons #10733: 6.0 Feedback: Protocol analysis with invalid Python API version fails without reporting to client

mcous, 2022-06-13:
### Overview

I you upload a Python protocol with a correctly shaped but invalid protocol API version (e.g. 2.42), the protocol analysis will fail on the server, but the analysis resource will not be updated, so the endpoint will report "pending" forever.

### Steps To Reproduce

This simple protocol triggers the bug:

```python

from opentrons import protocol_api

metadata = {
    "apiLevel": "2.1000",
}

def run(protocol: protocol_api.ProtocolContext) -> None:
    protocol.comment("hello world")
```

y3rsh, 2022-06-13:
Seems like a good candidate for an automated acceptance test?
1. POST the example protocol to produce the analysis failure.
2. Wait for analysis to complete
3. GET the analysis detail and validate status is correct and the right error is present.

ecormany, 2022-06-14:
I'd say expected behavior here should be a failed analysis with an error like you get with `opentrons_simulate`: 
```
RuntimeError: API version 2.1000 is not supported by this robot software. Please either reduce your requested API version or update your robot.
```

mikecardarelli, 2022-06-14:
There are designs for this error on the slide-out, but they haven't been implemented yet

mikecardarelli, 2022-06-14:
See #10569 for this issue

mcous, 2022-06-14:
Should not be closed, server bug

bndo, 2022-06-23:
Closed by #10780

closed 2022-06-23 by bndo