Opentrons/opentrons #10738: 6.0 Feedback:  On protocol run record start and end times are not displayed.

y3rsh, 2022-06-13:
### Overview

This protocol completed. The start and end times of each command are in the json when you download it, but are not shown in the app.
[image]

### Current Behavior

Not shown.

### Expected Behavior

Shown.

### Steps To Reproduce

Run a protocol with steps that complete.
cancel or complete teh run
from the devices screen  view the protocol run record
[image]

### Operating system

Mac

bndo, 2022-06-23:
No one has been able to repro this for a while. The related PR #10860 will be merged shortly

vabruzzo, 2022-06-24:
@bndo i've merged #10860 and unassigned myself from this ticket. I will move it to bug backlog in case it pops up again.

bndo, 2022-06-24:
I'm going to close this because it has the `required-for-6.0` label on it. @y3rsh just FYI for when you're back!

closed 2022-06-24 by bndo

nusrat813, 2022-06-29:
Verified fixed