Opentrons/opentrons #10782: 6.0 Feedback:  overflow menu is overlapping with slideout on device landing page

reachojhaneha, 2022-06-15:
### Current Behavior

[image]

### Steps To Reproduce

1. App is on 6.0.0-beta-0 
2. click on overflow menu
3. Click on run a protocol button
4. the slideout opens but the overflow menu is persistent.

b-cooper, 2022-07-01:
Closed by #10990

closed 2022-07-01 by b-cooper

y3rsh, 2022-07-06:
### verified alpha5
https://user-images.githubusercontent.com/502770/177564458-e42a3728-0df8-4059-9930-6e92fba6a205.mp4

commit b0f3a047 (2022-07-01): fix(app): remove slideout and modal from robot overflow menu conditional render

Prevent the robot overflow menu from visually persisting above the choose a protocol slideout when
it is opened. Also fix a bad icon svg path that was leaving errors in the console.

Closes #10782, Closes #10536
PR #10990 (2022-07-01): fix(app): remove slideout and modal from robot overflow menu conditional render
# Overview

Prevent the robot overflow menu from visually persisting above the choose a protocol slideout when
it is opened. Also fix a bad icon svg path that was leaving errors in the console.

Closes #10782, Closes #10536

# Changelog

- move the `ChooseProtocolSlideout` and `ConnectionTroubleShooting` modal out of the block that is shown or hidden with the robot overflow menu
- explicitly close the robot overflow menu if either of the above modal or slideout are opened
- fix up a bad svg icon path for `open-in-n [... cut]
commit 884396d9 (2022-07-01): fix(app): remove slideout and modal from robot overflow menu conditional render (#10990)

Prevent the robot overflow menu from visually persisting above the choose a protocol slideout when
it is opened. Also fix a bad icon svg path that was leaving errors in the console.

Closes #10782, Closes #10536