Opentrons/opentrons #10797: 6.0 Feedback:  Protocols with long names overlap in protocol run history

nusrat813, 2022-06-16:
### Overview

1315X840

### Current Behavior

[image]

### Operating system

Windows

koji, 2022-06-16:
- [ ] Need the design team support for spacing

koji, 2022-06-17:
Ask about adding padding to the protocol to Emily and Mel. If they agree with that as a solution, we can add it.
If not, we should close this since actually there is no overlap.

nusrat813, 2022-07-06:
@koji - any update on this?

koji, 2022-07-08:
@nusrat813 This will be solved by #11054

closed 2022-07-18 by koji