Opentrons/opentrons #10798: 6.0 Feedback (external):  Unable to find a labware definition in the protocol analysis

bndo, 2022-06-16:
### Overview

1. User receives a Protocol analysis failure message in the Protocols tab.

2. User is able to see the definition in the Labware tab

3. Error details convey that the labware definition is not found.

`FileNotFoundError [Line 39]: Unable to find a labware definition...`
[image]

### Steps To Reproduce

Protocols are available in Google Drive.

### Operating system

Windows

### Robot setup or anything else?

Attempt to reproduce internally was unsuccessful.

y3rsh, 2022-06-16:
Verified, this seems to be a Windows only issue.  On Mac and Linux custom labware that is imported is found.
Example protocol with 2 custom labware is attached for testing and dev.
[customLabwareProtocolWithAssets.zip](https://github.com/Opentrons/opentrons/files/8920793/customLabwareProtocolWithAssets.zip)

koji, 2022-06-16:
@y3rsh 
Windows version is 10, isn't it?