Opentrons/opentrons #10804: 6.0 Feedback: app no longer syncs robot time

mcous, 2022-06-16:
### Overview

Prior to the OT-2R, OT-2's did not ship with a real-time clock module installed. This means that whenever an OT-2OG is turned off, it loses track of the time.

To account for this, #6501 added functionality to the app that would sync the OT-2's clock with the user's computer when the robot was "connected". The v6 app seems to have removed this behavior, which will cause large and confusing time differences between non-refresh OT-2s and the app anywhere timestamps are used.

### Current Behavior

The app never checks or sets the OT-2's clock.

### Expected Behavior

At some point before a robot is used or interacted with, the app should:

1. Check the robot's time
2. If the robot's time disagrees by some threshold with the app, set the robot's time
    - <= v5 app uses 60 seconds as the threshold

### Robot setup or anything else?

See https://github.com/Opentrons/opentrons/blob/f003365aa47b3906f5ff6b3162e200d786043a5e/app/src/redux/robot-admin/epic/syncTimeOnConnectEpic.ts for previous sync time logic

bndo, 2022-06-23:
Connect to clock before the run is created.

mikecardarelli, 2022-06-27:
I think we can live without this for the next BETA release.

PR #10933 (2022-06-28): fix(app): sync robot clock on device and run detail page
# Overview

Instead of anchoring the system clock syncing off of the now orphaned robot connect action, create
an explicit sync time action and dispatch it on mount of the device and run details pages.

Closes #10804

# Changelog

 - keep existing epic logic in place, but change from `syncTimeOnConnectEpic` to `syncSystemTimeEpic`. instead of listening for the `robot:CONNECT` action, this epic now listens for a new purpose-built action called `robotAdmin:SYNC_SYSTEM_TIME`
 - make a hook called `useSyncRobotClock` that syncs the robot clo [... cut]
commit 06a73a8c (2022-06-29): fix(app): sync robot clock on device and run detail page (#10933)

Instead of anchoring the system clock syncing off of the now orphaned robot connect action, create
an explicit sync time action and dispatch it on mount of the device and run details pages.

Closes #10804
closed 2022-07-01 by y3rsh