Opentrons/opentrons #10828: 6.0 Feedback: a corrupted or invalid protocol analysis prevents new protocols from being opened

mcous, 2022-06-17:
### Overview

The app's protocol analysis handling has some brittleness that can fundamentally break the UX. If an invalid protocol analysis file ends up in the app's protocol directory:

- Other (valid) protocols will not be listed
    - Instead, the UI will look like the app has _no_ stored protocols
- The UI will be completely unresponsive if the user tries to open a new protocol
    - The new protocols will still be added to the stored protocols directory in the background

Discovered while testing #10827. The only cause on an invalid analysis JSON file would be a bug in our own software, but our software has bugs sometimes 🙈 

### Current Behavior

An invalid protocol analysis in the `protocols` directory prevents any protocols from being listed.

### Expected Behavior

- Valid protocols continue to show up
- Invalid protocols either
    - Don't show up
    - Shop up with an "unable to analyze" warning, similar to an analysis error

### Steps To Reproduce

1. Upload a protocol
2. Go to your app's protocols directory
    - macOS: `open ~/Library/Application\ Support/Opentrons/protocols`
3. Find an analysis JSON file and make it invalid JSON in some way
4. Launch the app

b-cooper, 2022-07-01:
Closed by #10969

closed 2022-07-01 by b-cooper

y3rsh, 2022-07-01:
I deleted the starting brace in an analysis file of a protocol.  I then Force reloaded and looked at the analysis error for the protocol with no issues.  All other protocols were also available on the protocols view.

PR #10969 (2022-07-01): fix(app-shell): allow corrupt protocol analysis files to error gracefully
# Overview

Return all stored protocols from the file system, even if some failed to have their analysis parsed.

Closes #10828

# Changelog

- wrap analysis JSON parsing in try catch and construct failed analysis if parsing fails

# Review requests

- [ ] alter an analysis json by hand to make it invalid JSON and ensure that you can still access protocols on the Protocols Tab 

# Risk assessment
low
commit ed72dc30 (2022-07-01): fix(app-shell): allow corrupt protocol analysis files to error gracefully (#10969)

Return all stored protocols from the file system, even if some failed to have their analysis parsed.

Closes #10828