Opentrons/opentrons #10831: If robot needs to be updated, always display the update available banner and warn if robot is busy

emilywools, 2022-06-17:
1. If the robot has an update available, always show the software update available banner on the robot card on the robot landing page, the robot detail page and the Robot Settings > Advanced Settings page

2. If the user launches the View Update modal and clicks Update Robot, show an “Are you sure” modal to the user to let them know updating will kill any in-progress actions.

See Figma for design details: https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=5196%3A190443

closed 2022-06-23 by bndo

y3rsh, 2022-06-30:
We **STRONGLY** suggest you update your robot.

https://user-images.githubusercontent.com/502770/176705860-fc06516d-65a5-4ec5-9b53-6dc82d89ca4d.mp4

Confirmation modal present.

https://user-images.githubusercontent.com/502770/176706973-d774d2b7-5bd8-4373-add6-a7a51d043ae8.mp4

emilywools, 2022-06-30:
maybe you shouldn't be able to dismiss this banner and we should make this more emphatic.

A software update is available for this robot. <something about how you need to update to access the robot's functionality>.

cc @ecormany ?

PR #10732 (2022-06-22): fix(app): wire up Software Update Alerts toggle logic
closes #10576, closes #10783, closes #10831

# Overview

When you switch `Software Update Alerts` toggle to off, this PR fixes it so you won't receive the modal popup when there is an update available. It also removes the `useIsRobotBusy` hook from the update banners which is in the 1) Robot Card 2) Robot overview and 3) Robot Software Version. 

Additionally, this PR fixes the  X icon button on the banner. It will dismiss when you click on it now.

Also, if you try to update your robot when the robot is busy, there will be a warning modal
 [... cut]