Opentrons/opentrons #10839: 6.0 Feedback: Heater-Shaker Module controls (card, slide-out, modal)

ecormany, 2022-06-17:
### Overview

A variety of unexpected behaviors when controlling a module connected to an OT-2 from the Device Detail page.

### Current Behavior

1. Incorrect statuses when turning on module:
	1. Heater listed as "Cooling" (persists until temperature set). Relatedly, first item in overflow menu starts as "Deactivate". Should begin as "Set target temperature" and change to "Deactivate heater".
	2. Shaker listed as "Holding At Target" (changes to "Idle")

https://user-images.githubusercontent.com/1958332/174400383-8b6815b6-090f-41ca-9529-5708245128fd.mov

2. Target speed shown in card can be different than input target. Specifically, it's always rounded down to a multiple of 6. This is most noticeable when setting the shake speed to 200, which is rounded down and reported as 198, which is below the minimum target allowed.
4. Slide-outs: hitting enter does not submit the target temperature or speed.
5. Slide-outs: input value is persisted *across* temperature and shake slide-outs. This is not useful, since the two acceptable ranges don't overlap.

https://user-images.githubusercontent.com/1958332/174400546-2dc95384-a477-4baf-bc1f-713285ff72a4.mov

5. Slide-outs: `-`, `--`, and `--` + an in-range value are all accepted inputs but result in no module action.
6. Confirm attachment modal: background dim is misaligned

https://user-images.githubusercontent.com/1958332/174401044-5e0357a9-0927-4d60-bad9-8ffc1a9e10f9.mp4

[image]

7. Copy: no space between numbers and "°C", e.g. "37°C"

### Expected Behavior

1. Module should be reported as "Idle" for both heater and shaker when powered on.
2. Target speed should be exact value entered by user (or rounded to nearest 1 rpm).
3. Hitting enter in the text field should have the same effect as clicking the "Confirm" button.
4. Input values should be persisted separately for the temperature and shake slide-outs.
5. Values beginning with `-` should be rejected.
6. Dimming rectangle should be aligned to window edge.
7. Should be a space between number and "°C", e.g. "37 °C".

### Operating system

Mac

### Robot setup or anything else?

App on `release_6.0.0` branch commit [0e8888d](https://github.com/Opentrons/opentrons/commit/0e8888d2eb066a1432b56799796701f41c17890e).

Heater-Shaker on firmware v0.4.4-0b068f7

ecormany, 2022-06-17:
This is a lot, so feel free to break down into smaller issues.

jerader, 2022-06-22:
These 2 AC:

- Input values should be persisted separately for the temperature and shake slide-outs.
 - Values beginning with - should be rejected.
 
 will be addressed in a new ticket: https://github.com/Opentrons/opentrons/issues/10865

jerader, 2022-06-28:
Module should be reported as "Idle" for both heater and shaker when powered on. 

- this bug is a firmware issue and was addressed in the latest internal release version: 0.4.5

commit e9e66622 (2022-06-24): fix(app): module cards and slideouts various bug fixes

closes #10839 and #10854
commit 6180f173 (2022-06-24): fix(app): module cards and slideouts various bug fixes

closes #10839 and #10854
PR #10864 (2022-06-28): fix(app): module cards and slideouts various bug fixes
partially closes #10839 and fully closes #10854 and #10856

# Overview

This PR addresses various small fixes within the module cards and module slideouts:

1. fix some spacings
2. slideout buttons are clickable via the enter button on keyboard
3. confirm attachment modal is now rendered above breadcrumbs
4. input field no longer accepts decimal points or negative/positive symbols
5. module commands are accessible via the robot overview module cards when the run is in terminal state (failed, succeeded, stopped)
6. module commands are acces [... cut]
commit 7b90a79c (2022-06-28): fix(app): module cards and slideouts various bug fixes (#10864)

partially closes #10839 and fully closes #10854 and #10856