Opentrons/opentrons #10841: issue: ssh from python, correct slot but wrong position for tip pick up, labware offset maybe not working

m89ozturk, 2022-06-21:
I have written a very simple protocol where 20ul and 300ul pipettes pick up tip and return to original positions.
The labware offset data is included in the protocol and works fine when manually connected with ssh and executed with opentrons_execute.
However, it fails to position correctly and pick up the tip when the same protocol is run through ssh connection established in python (subprocess.Popen) even though the steps of the protocol runs. Therefore, to me it seems the offsetting of the labware is not working.
Any help would be greatly appreciated.

CODE:
# How the protocol is run
mycmd = "RUNNING_ON_PI = 1; opentrons_execute /var/lib/jupyter/notebooks/simpleProtocol_ssh.py"
print(mycmd)
proc = subprocess.Popen(f"ssh -i ot2_ssh_key {user}@{host} {mycmd}", shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
result = proc.stdout.readlines()
print(result)

# Protocol
from opentrons import protocol_api
metadata = {'apiLevel': '2.12'}
def run(protocol: protocol_api.ProtocolContext):
   tiprack1 = protocol.load_labware('opentrons_96_tiprack_20ul', 10)
   tiprack1.set_offset(x=-0.80, y=1.20, z=0.00)
   p20_single = protocol.load_instrument('p20_single_gen2', 'left' , tip_racks = [tiprack1])
   p20_single.pick_up_tip()
   p20_single.return_tip()

   tiprack2 = protocol.load_labware('opentrons_96_tiprack_300ul', 11)
   tiprack2.set_offset(x=-0.10, y=1.00, z=0.20)
   p300_single = protocol.load_instrument('p300_single_gen2', 'right' , tip_racks = [tiprack2])
   p300_single.pick_up_tip()
   p300_single.return_tip()

   protocol.home()

mcous, 2022-06-22:
Not sure if it'll fix the issue, but your SSH command definitely has a couple typos that will prevent the `RUNNING_ON_PI` environment variable from being set correctly. You should remove the spaces around the `=` sign and get rid of the semicolon.

```python
mycmd = "RUNNING_ON_PI=1 opentrons_execute /var/lib/jupyter/notebooks/simpleProtocol_ssh.py"
```

m89ozturk, 2022-06-23:
oh wow yes, works perfectly. thank you.

closed 2022-06-23 by m89ozturk