Opentrons/opentrons #10874: 6.0 Feedback: 'No module named pandas'

nusrat813, 2022-06-23:
### Current Behavior

Pandas package not found when uploading protocol 

slack nus for protocol 

reproducible on mac and windows 
[image]

### Operating system

Windows

SyntaxColoring, 2022-06-23:
This looks like an instance of the app's local Python environment differing from the robot's Python environment.

`pandas` is on the robot I'm *pretty sure,* and protocols may expect it to be there. It's almost certainly not in the app's local Python environment.

ecormany, 2022-06-23:
`pip list` on my robot running 6.0.0a2 lists pandas 0.25.3. I've never installed it on my local Python.

closed 2022-06-29 by nusrat813

y3rsh, 2022-06-29:
verified
[smalles.py.zip](https://github.com/Opentrons/opentrons/files/9011072/smalles.py.zip)
is analyzed on the app.

commit dc314970 (2022-06-28): build(app-shell): build app's python env with pandas as dependency to match robot's env (#10890)

Include the `pandas` library in the app's python environment as it is a default package available on
the OT2's python environment.

Closes #10874