Opentrons/opentrons #10888: 6.0 Feedback: Design Feedback -  FactoryReset Slideout

koji, 2022-06-23:
- [x] (1) Up the size of this icon to 24px, currently it's 20px
 - [ ] (2) Wondering why this is a different style from design?
 - [ ] (3) Add 8px margin between the list items
 - [ ] (4) Looks like there's a barely noticeable grey line right underneath the button?
 - [ ] (5) remove box shadow from .layXKe
 - [ ] (6) Looks like the SVG is a bit thicker than the designs—when you get to this step, contact me and I can try sending you another SVG
 - [ ] (7) The hover state colour should be our darkGreyHover (#646468), currently it's #283d52
 - [ ] (8) When I try to tab through all the links using my keyboard none of the links of the tabs go into their focusVisible state

[image]

The original feedback is in #10819

koji, 2022-06-24:
@mmencarelli 
Could you tell me how you find No.4 line? (I don't see it).
Could you tell me about No.5 and No.6 since they are not in the image you shared with us?

mmencarelli, 2022-06-24:
> @mmencarelli Could you tell me how you find No.4 line? (I don't see it). Could you tell me about No.5 and No.6 since they are not in the image you shared with us?

1. I've just checked again and it's still there! but it's just very very faint. it's exactly here at the bottom of this blue space 
[image]

2. I can't seem to find where i'm referring to removing the box-shadow, you're good for now! 

Also, I'm noticing there's no margin between the checkbox and the copy. Don't know if this is in progress, but just make sure that gets back in there!

koji, 2022-06-24:
@mmencarelli 
Thank you for pointing out the space issue (I removed it accidentally)

PR #10894 (2022-06-28): style(app): Fix design feedback on FactoryReset

# Overview
This will fix design feedback on FactoryReset and update CheckboxField styling.
fix #10888

# Changelog
- Update FactoryReset Slideout styling
- Update CheckboxField styling
- Add scroll check to Slideout

# Review requests
- The UI aligns with the design #10888

# Risk assessment
low
commit 77d35fc8 (2022-06-28): style(app): Fix design feedback on FactoryReset (#10894)

This will fix design feedback on FactoryReset and update CheckboxField styling.
fix #10888
closed 2022-06-29 by koji