Opentrons/opentrons #10906: 6.0 Feedback: [Design QA] T2 Protocol Detail

mmencarelli, 2022-06-24:
### Overview

**This is not a full design QA for T2 Protocol Detail page. Due to time constraints, I'm only including what needs to be done for 6.0**

### Expected Behavior

[Link to annotated frame in Figma ](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=14699%3A293314)

- [ ] (1) Update `Card__Section-r4iqug-0 itnElk` to reflect designs. Remove box-shadow, add border, border-radius 
- [ ] (2) Update `Card__Section-r4iqug-0 bopoGf` to reflect designs. Remove box-shadow, add border, border-radius 
- [ ] (3) Update to reflect 'h2-semibold' styling 
- [ ] (4) Remove box-shadow from button
- [ ] (5) Wrap text in the org/author section. This column is the same width as 'creation method' column above. That way 'description' left-aligns with 'last updated' above 
- [ ] (6) Push all these eyebrows to darkGreyEnabled
- [ ] (7) Missing the hover state on this tab. I know someone else is working on this, so please align! 
- [ ] (8) Are we supposed to have this button be enabled? In the designs it isn't. 
- [ ] (9) Fix the spacing on the open menu. Needs to be 4px below and right aligned with the button that opened it 
- [ ] (10) Make sure it's .5rem spacing between headers and the divider beneath it 
- [ ] (11) When you click 'read more', the font size for copy under "description" changes. [Loom video here. ](https://www.loom.com/share/1d258464c9a14871883a6feb1751df4a)

[image]
[image]
[image]

mmencarelli, 2022-06-28:
> @mmencarelli, for this item: (8) Are we supposed to have this button be enabled? In the designs it isn't. - it is supposed to be enabled, it is a way to upload a protocol and go to protocol setup. I'm not sure if there is a disabled reason attached to it though.

Just going back into this and the reason why I thought it would be disabled is that this is the _loading state_ of the protocol page. Just want to confirm that we still allow the user to go to the protocol run page from a loading state of the protocol detail page!

jerader, 2022-06-28:
> Just going back into this and the reason why I thought it would be disabled is that this is the _loading state_ of the protocol page. Just want to confirm that we still allow the user to go to the protocol run page from a loading state of the protocol detail page!

@mmencarelli , i see! I missed that the image is showing the loading state. Yes, I assume we do want it to be disabled. I can add that.

PR #10932 (2022-07-02): refactor(app): protocol details page design qa feedback 1
closes #10906

# Overview

address design feedback on Protocol Details page: 

[image]

# Changelog

- changed `ProtocolDetails` and `ProtocolLabwareDetails`

# Review requests

- upload a protocol and click on the protocol card to get the protocol labware details. does it match designs?

# Risk assessment

low
commit becf429e (2022-07-02): refactor(app): protocol details page design qa feedback 1 (#10932)

closes #10906