Opentrons/opentrons #10926: 6.0 Feedback: [Design QA] Protocols T1 Slideout

koji, 2022-06-27:
### Overview
The original feedback is in #10618
[To a dev who picks this up]
Please do not close #10618 until finish fixing all slideouts design feedback.
If you have any questions on this feedback, please ask about your questions to Mel.

#### * Slideout close icon size has been changed into `24px` so you don't need to modify it.

### Current Behavior

Link to [annotated frame in Figma ](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=13793%3A416275)

### **Import custom labware def + toast**
- [x] (1) Size of this button should be 24px (currently 20px)
- [ ] (2) remove the box-shadow
- [ ] (3) Going to sneak this toast edit in here for simplicity sake, but can we add a max-width of these toasts to like 70% of the viewport? This percentage should also work when we shrink down to our min-width size 
- [ ] (4) Can we make it a rule that the toasts stay for 8s instead of 4? Noticing that it's a bit quick. 
- [ ] (5) Could we have the toast fade in and out like[ this example? ](https://polaris.shopify.com/components/feedback-indicators/toast#navigation)
- [ ] (6) Center this text
- [ ] (7) needs 2rem margin-top

[image]
[image]

koji, 2022-07-01:
@mmencarelli 
We can add a similar animation to Toast. What we can do is the following.

1. switch animation from `slide-in` to `slide-up`.
2. keep both animations (`slide-in` and `slide-up`) and use one of animations depending on a page

mmencarelli, 2022-07-05:
> 

This sounds good, would it be possible to check on this branch what you're referring to?

koji, 2022-07-05:
> > 
> 
> This sounds good, would it be possible to check on this branch what you're referring to?

@mmencarelli yes, if you check out this branch and try to import a protocol into the app.
You can see slideup animation and the current duration is 5 sec.

PR #10994 (2022-07-08): fix(app): Address Labware import slideout design feedback

# Overview
This will address labware slideout design feedback and update Toast functionality
fix #10926

* In terms of Toast, got confirmation from Mel
- use slideup animation for our Toast
- use 5 sec as the default duration

sample: https://polaris.shopify.com/components/toast#navigation

# Changelog
- address design feedback
- add `duration` to Toast and set the default duration 5s 
- add a new test case for `duration`

# Review requests
- design feedback is addressed
- Would like to get feedba
commit 6943c37e (2022-07-08): fix(app): Address Labware import slideout design feedback (#10994)

this will address labware slideout design feedback and update Toast functionality
fix #10926
closed 2022-07-15 by b-cooper