Opentrons/opentrons #10935: Block protocol run when a robot and app are not on the same SW version

emilywools, 2022-06-28:
When the Opentrons App and an OT-2 are not on the same SW version, disable running a protocol in the following places:

1. Device landing page: on a robot card, disable `Run a protocol` from the overflow menu
2. Device detail page: disable `Run a protocol` on the upper right corner
3. Device detail page: disable `Rerun protocol now` from the overflow menu
4. Run detail page: disable `Run again` on the upper right corner

In all of these scenarios, display a tooltip that says 
> "A software update is available for this robot. Update to run protocols."

SyntaxColoring, 2022-06-28:
This might need care to ensure that we can still test our changes during development. During development, it's common for versions to not match exactly.

For example, if the app is on `vX-alpha.1` and I'm testing a robot-side PR that's based on `vX-alpha.0`, I think the app currently counts that as a mismatch. (It prompts me to upgrade my robot.)

closed 2022-07-08 by vabruzzo

y3rsh, 2022-07-12:
[image]

[image]

[image]

[image]

PR #11050 (2022-07-08): fix(app): disable protocol runs when robot update available
# Overview
When robot update is available disable run protocol buttons.
Closes #10935.

# Changelog
- disables device landing page robot card run a protocol in the overflow menu
- disables device detail page run a protocol button
- disables device detail page rerun protocol now from the overflow menu
- disables run detail page run again button
- adds tooltip to each of these

# Review requests
Confirm that these items are disabled when robot software update is available

# Risk assessment
Medium, this affects potentially blocking runs [... cut]
commit 3618c9a7 (2022-07-08): fix(app): disable protocol runs when robot update available (#11050)

Closes #10935.