Opentrons/opentrons #10939: 6.0 Feedback:  Tip length calibration method selection ignored before calibration beings

nusrat813, 2022-06-28:
### Current Behavior

[image]

[image]

When a user selects always show prompt for calibration method, user is not prompted with modal and automatically gets trash bin as calibration method 

### Expected Behavior

User should get prompted with selection modal 

### Operating system

Mac

nusrat813, 2022-06-28:
Not a beta blocker but we want to get it into release

y3rsh, 2022-07-01:
I am not being prompted.

https://user-images.githubusercontent.com/502770/176953963-ba57fdfe-276e-4ffd-9420-24b7326c3186.mp4

brenthagen, 2022-07-01:
@y3rsh are there any other ways to launch tip length calibration other than from:
 - the protocol run details page setup tab, addressed in https://github.com/Opentrons/opentrons/pull/10976
 - the robot settings page calibration tab, addressed in https://github.com/Opentrons/opentrons/pull/10991

closed 2022-07-05 by nusrat813

y3rsh, 2022-07-06:
verified.  The settings seem to be respected everywhere I opened tip calibration from
robot device details
robot settings

PR #10976 (2022-06-30): fix(app): fix tip length calibration method selection
# Overview

fixes bug where tip length calibration method selection in app settings was ignored on tip length calibration modal
launch

[image]

closes #10939

# Changelog

 - Fixes bug where tip length calibration method selection in app settings was ignored on tip length calibration modal
launch

# Review requests

confirm that app settings tip length calibration method selection is reflected - the selection "Always show the prompt to choose calibration block or trash bin" should display the block selection modal on tip length calibration [... cut]
commit 9d4bec01 (2022-06-30): fix(app): fix tip length calibration method selection (#10976)

fixes the bug where tip length calibration method selection in app settings was ignored on modal
launch

closes #10939