Opentrons/opentrons #10942: 6.0 Feedback:  Setup for run: Deck map labware offsets seem unaligned

y3rsh, 2022-06-29:
### Overview

On the protocol setup page complete LPC and then see the deckmap has the offsets but they all look a little off.
[image]

emilywools, 2022-06-29:
hey Josh do you mean they look off visually or the values are wrong?

y3rsh, 2022-06-30:
I believe they are accurate values, they are off visually.  The `T` text in the row is misaligned vertically and then in some places the `T` is even missing or is chopped off.

closed 2022-07-07 by smb2268

rwarshauer, 2022-07-12:
Issue is resolved in Beta2
[image]

emilywools, 2022-07-12:
@rwarshauer this does not look correct. the labware in slots 1, 2, 3, 5, 6 has the labware name right aligned instead of left aligned.

rwarshauer, 2022-07-12:
[image]
Beta2 
Alignment of some slots is to the left and some to the right as mentioned by Emily

commit ebe2ed55 (2022-07-07): fix(app): fix labware info overlay styling with vector (#11008)

fix #10942