Opentrons/opentrons #11042: 6.0 Feedback:  Robot is not homing after canceling our of 'Detach pipette'

rwarshauer, 2022-07-07:
### Overview

After canceling a detach pipette command the robot lifts the pipettes but does not home the robot

### Current Behavior

Robot does not home

### Expected Behavior

Robot homes

### Steps To Reproduce

1. Click on robot
2. Under robot click pipette
3. Click detach
4. Click exit after pipette comes to fron
[image]
t

### Operating system

Mac

rwarshauer, 2022-07-12:
Issue is also seen when successfully detaching pipette as well. Robot does not home.

nusrat813, 2022-07-12:
To Do: Check if it is a regression

bndo, 2022-07-13:
This is in 5.0.2, let's consider this in the new redesigned flows for a next release