Opentrons/opentrons #11060: 6.0 Feedback:  Update error message for busy robot in choose a robot slideout

emilywools, 2022-07-08:
### Overview

**current behavior:**
user is on protocol detail page and clicks "run protocol"
user chooses a robot that's busy and clicks "Proceed to setup"
user sees this error on the robot card:

> Current run is not idle or stopped

[image]

**desired behavior:**
user is on protocol detail page and clicks "run protocol"
user chooses a robot that's busy and clicks "Proceed to setup"
user sees this error on the robot card (see figma):

> this robot is busy and can't run this protocol right now. <Go to robot>

https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=15319%3A295635

nusrat813, 2022-07-12:
verified in beta.2 [image]

emilywools, 2022-07-12:
hey @nusrat813 is that the right screenshot? the error should not say "current run is not idle or stopped." it should say "this robot is busy and can't run this protocol right now. Go to robot"

commit 0a134ab4 (2022-07-11): fix(app): change error text when creation error is run is busy

closes #11060
commit b9f2ba44 (2022-07-11): fix(app): change error text when creation error is run is busy (#11069)

closes #11060
PR #11106 (2022-07-12): feat(app): fix choose protocol/robot slideout robot busy error logic
closes #11060

# Overview

fix logic in ` useCreateRunFromProtocol` to look for `protocolError?.response?.status ?? runError?.response?.status` instead of `runError?.code`

# Changelog

- change prop and logic in ` useCreateRunFromProtocol` and update usage in slideouts

# Review requests

- start protocol on robot and pause protocol. Try to upload another protocol and click `proceed to setup`. You should get correct error 

# Risk assessment

low
commit d7ae58bb (2022-07-12): feat(app): fix choose protocol/robot slideout robot busy error logic (#11106)

closes #11060