Opentrons/opentrons #11061: 6.0 Feedback: LPC: Add space between the xyz gray box and "need to make an adjustment" text

emilywools, 2022-07-08:
### Overview

if the "make an adjustment" text needs to wrap to 2 lines on LPC, add space between the gray box and the "make an adjustment" text.

current state:
[image]

design:
[image]

koji, 2022-07-15:
@emilywools @mmencarelli 

When users open LPC with the width of more than 1010px, actually the app has space between the offset and the text.
But at less than 1000px, the app loses space.

https://user-images.githubusercontent.com/474225/179288469-3c3c31fe-682f-4cfd-81af-ac4107198e4b.mov

I guess the ideal solution would be shrinking text that we don't have. (In the future we would be able to have since we talked about that a little bit in the design <> dev meeting on Thursday)

I think a simple workaround would be adding margin-right and margin-left to the text `make an adjustment?`

add 8px margin and the image's window width is 825px
[image]

Any thoughts?

koji, 2022-07-18:
@emilywools @mmencarelli 

This keeps the same white-background size as well as the above content.
[image]

[image]

This is almost the same as the first one. The difference is no bottom margin.
[image]

[image]

emilywools, 2022-07-18:
imo this is good for an interim solution (until we reskin LPC!)

koji, 2022-07-19:
This has been closed the above PR

closed 2022-07-19 by koji