Opentrons/opentrons #11067: bug: cannot dispense to a tiprack

silvtal, 2022-07-11:
# Overview

Before 2.11, I was able to define a custom "tip_isolator", fill it with water and rest reusable tips there. Now I cannot fill it with water and I have to do it manually

```
    # Define tip_isolator
    tip_isolator = ctx.load_labware(tiprack_type[tip_type], 11, 'Tip Isolator')

    # Add 300 uL of Water to each well in tip isolator
    ctx.comment('''Transferring 300 uL of water to
                each well in the tip isolator''')
    pick_up(m300, tip_isolator['A1'])
    m300.transfer(300, [well for well in res2.wells()[9:12]
                        for _ in range(4)][:cols],
                  tip_isolator.rows()[0][:cols], new_tip='never')
    m300.drop_tip(_drop(m300))
```

# Steps to reproduce

```
    # Define tip_isolator
    tip_isolator = ctx.load_labware(tiprack_type[tip_type], 11, 'Tip Isolator')

    # Add 300 uL of Water to each well in tip isolator
    ctx.comment('''Transferring 300 uL of water to
                each well in the tip isolator''')
    pick_up(m300, tip_isolator['A1'])
    m300.transfer(300, [well for well in res2.wells()[9:12]
                        for _ in range(4)][:cols],
                  tip_isolator.rows()[0][:cols], new_tip='never')
    m300.drop_tip(_drop(m300))
```

# Current behavior

Error pops out
```
RuntimeError [line 426]: Cannot dispense to a tiprack
```

# Expected behavior
I would like to be able to dispense water into my custom tipracks if I need to

mcous, 2022-07-11:
Dispensing liquid to a tip rack is an error in a majority of protocols (see #7552), so I've reclassified this as a "feature request" rather than a bug. Disallowing aspirates/dispenses with tip-racks is fully intended behavior of the Protocol API.

We'll have to consider this idea of a "hybrid" labware that can be a receptacle for both liquids and tips a little carefully. In the mean time, you have a few options for workarounds:

- Keep your protocol locked to the PAPI version you noted (`2.10`)
    - This has the downside of locking you out of unrelated bug fixes in future PAPI versions
- Strip the labware information from the dispense location to bypass the tip-rack check
    - This may not work in some future PAPI version, but should hold up for now

This little snippet shows you how to strip labware information from a given well location:

```python
from opentrons import types

...
tip_well = tip_rack.wells()[0]
tip_well_dispense_location = types.Location(
    point=tip_well.bottom(z=pipette.well_bottom_clearance.dispense).point,
    labware=None,
)
...

pipette.dispense(volume=20, location=tip_well_dispense_location)
```

silvtal, 2022-07-12:
Maybe a solution would be to throw an error when the tiprack is an Opentrons tiprack, but only throw a warning when the tiprack is custom? 

Thank you for the workarounds you proposed. I didn't want to be stuck with version 2.10, but the second option seems good