Opentrons/opentrons #11096: 6.0 Feedback: LPC not handling error thrown when Thermocycler lid is closed

DerekMaggio, 2022-07-12:
### Overview

When I have a protocol with a Thermocycler in it and the lid is closed, when I perform LPC it throws an unhandled error. When I manually open the lid and rerun LPC I do not get the error.

### Current Behavior

See overview

### Expected Behavior

Given you are running LPC and the Thermocycler lid is closed.
When I hit the step for LPC to check the Thermocycler 
I should get an error saying the lid is closed.
The error should direct me to open the lid. 
The error should have a button to continue LPC.
The system should verify that the lid is open and move on with LPC. 
If the lid is still closed the error should show again

### Steps To Reproduce

1. Load a protocol that has a thermocycler in it
2. Make sure the thermocycler lid is closed
3. Run LPC
4. When you hit the step to run LPC for thermocycler you will get an error

https://user-images.githubusercontent.com/20705756/178536687-09fdebd8-0c39-4157-b24f-8befc361def3.mp4

### Operating system

Linux

y3rsh, 2022-07-12:
The error is due to a 410 from
POST to
http://192.168.50.89:31950/modules/TCV0220211102A02
{"command_type":"open","args":[]}

response is 
{"message":"This endpoint has been removed. Use POST /commands instead."}

bndo, 2022-07-12:
Closed by #11111

PR #11111 (2022-07-12): fix(app): open tc lid in LPC via protocol engine command
# Overview

This PR issues a TC open lid command (rather than using the legacy modules endpoint which does not work with new version headers) to close the TC lid at the beginning of LPC.

Closes #11096

# Changelog

- Open TC lid in LPC via Protocol Engine

# Review requests
Make sure LPC works with a TC

# Risk assessment

Low
commit f74853b0 (2022-07-12): fix(app): open tc lid in LPC via protocol engine command (#11111)

Closes #11096
closed 2022-07-12 by bndo