Opentrons/opentrons #11101: 6.0 Feedback:  Robot disappears after changing name

rwarshauer, 2022-07-12:
### Overview

When robot name is change robot disappears 

### Current Behavior

Robot disappears 

### Expected Behavior

Robot goes unavailable then becomes available again

### Steps To Reproduce

1. Go to app
2. Go to robot settings
3. Go to advance
4. Change name
5. Observe

### Operating system

Mac

### Robot setup or anything else?

Robot only becomes usable again when plugged in

Uploading Screen Recording 2022-07-12 at 1.43.24 PM.mov…

rwarshauer, 2022-07-14:
[Untitled 2.mov.zip](https://github.com/Opentrons/opentrons/files/9113586/Untitled.2.mov.zip)

rwarshauer, 2022-07-14:
@koji Recording uploaded. Couldnt recored the whole time or it got to long, but going on 8 mins still say unavailable

koji, 2022-07-18:
- [ ] specify the condition that cause this issue

SyntaxColoring, 2022-07-22:
I think this bug has something to do with whether the "Unavailable" section contains any devices other than the one being renamed.

* **When there are no other unavailable devices:** the renamed robot is momentarily unavailable, and then it becomes available after a few seconds.
* **When there is another unavailable device:** the renamed robot disappears from both the "Available" and "Unavailable" sections. It never reappears in either of them until the app is restarted or the page is refreshed with **Cmd+Shift+R**.

Steps to reproduce:

1. Start off with a clean 6.0.0 app. Clear all devices from the "Unavailable" list. Perhaps also restart the app for good measure.
2. Join a Wi-Fi network with a single device on it.
3. Try renaming that device a bunch of times. It will come back each time.
4. Get some other device to show up in the "Unavailable" list. The way I did this was to:
    1. Add `localhost` to the list of additional IP addresses to poll, in the app settings. 
    2. Run `make -C robot-server dev`.
    3. Wait for the `opentrons-dev` device to show up in the Opentrons App.
    4. Stop the `robot-server` dev server with `^C`.
    5. Wait for the `opentrons-dev` device to be moved to the "Unavailable" section in the Opentrons App.
5. Try renaming the original device again. It will disappear and never reappear, unless you restart the app or refresh the page with **Cmd+Shift+R.**

Here is a video of that:

https://user-images.githubusercontent.com/3236864/180482456-fc96567a-4348-44b2-a5b3-a62419cb42a2.mov

koji, 2023-05-11:
this issue is tracked by a jira ticket

closed 2023-05-11 by koji