Opentrons/opentrons #11109: 6.0 Feedback:  Protocol run record start and end times not shown

y3rsh, 2022-07-12:
### Overview

For this protocol start and end times are not shown for any step.
[image]

### API run responses
[responses.log](https://github.com/Opentrons/opentrons/files/9096507/responses.log)
### client protocol folder 

> **Hunch** Maybe this is happening because there are 2 local analyses?  The first analysis fails because the custom labware file is not imported yet.  This is a part of the smoke test on purpose to use the reanalyze feature.

[dbc62979-b87e-4790-9bc6-d8eaba722457.zip](https://github.com/Opentrons/opentrons/files/9096516/dbc62979-b87e-4790-9bc6-d8eaba722457.zip)

nusrat813, 2022-07-12:
@b-cooper - to investigate

y3rsh, 2022-07-12:
### The run log download
[kansas_16e1af36-159a-4c54-96d7-9e49b94d6df2_2022-07-12T13_40_54.874Z.json.zip](https://github.com/Opentrons/opentrons/files/9096969/kansas_16e1af36-159a-4c54-96d7-9e49b94d6df2_2022-07-12T13_40_54.874Z.json.zip)

b-cooper, 2022-07-13:
Closed by #11114

PR #11114 (2022-07-13): fix(app): fix labware offset text alignment, fix historical run log missing timestamps
# Overview

Ensure that labware names are left aligned in the LabwareInfoOverlay, even if an offset is present.
Ensure that historical runs which involved more than 60 LPC (or otherwise pre-play) run commands are
able to correctly glean the first post play run command in the absence of a current command key.
Remove unnecessary polling of the protocol record from the run detail page.

Closes #11109, Closes #11108

# Changelog

- fix text alignment on labware info overlay so that all text is left aligned
- han [... cut]
closed 2022-07-13 by b-cooper