Opentrons/opentrons #11228: App run log: support magnetic module commands

shlokamin, 2022-07-26:
# Overview

Now that JSON v6 commands are first class citizens (both in analysis and as run commands), we need our run log to parse them and display them properly. 

Previously, we were relying on  the `legacyCommandText` we got back from the server to hold the text to render in the run log, but now we'll have to parse he actual command and format the string ourselves.

# Implementation details
 
See `app/src/organisms/Devices/ProtocolRun/StepText.tsx` and implement cases for the magnetic module commands:
- `magneticModule/engage` -> should map to text "Engaging Magnetic Module"
- `magneticModule/disengage` -> should map to text: "Disengaging Magnetic Module"

# Acceptance criteria

Magnetic module commands render text above in run log

ecormany, 2022-07-27:
- `temperatureModule/waitForTemperature` -> could be shortened to "Waiting for Temperature Module to reach ~~temperature~~ {temp} °C (rounded ~~off~~ to nearest integer)"
    - and {temp} should actually be rounded in display! currently it will say something like `50.0 °C (rounded…` 🙃 
- should `thermocycler/waitForBlockTemperature`, `thermocycler/waitForLidTemperature`, and `heaterShaker/waitForTemperature` list what the target {temp} is? or can they not because they don't take an argument?
- `thermocycler/setTargetBlockTemperature` ->  "Setting Thermocycler ~~well~~ block temperature to {temp} °C with a hold time of {seconds} seconds" (to match wording in other TC block actions)
- `heaterShaker/deactivateHeater` -> should map to text "Deactivating ~~Heater-Shaker~~ Heater"
- `heaterShaker/deactivateShaker` -> should map to text "Deactivating ~~Heater-Shaker~~ Shaker"

jerader, 2022-07-27:
> * should `thermocycler/waitForBlockTemperature`, `thermocycler/waitForLidTemperature`, and `heaterShaker/waitForTemperature` list what the target {temp} is? or can they not because they don't take an argument?

@ecormany yup, I do have access to what the target temp is

ecormany, 2022-07-27:
No problem, write the best text with what you have, this shouldn't have to be a big data pipeline refactor.

commit 083975b1 (2022-07-27): feat(app): support schemaV6 commands in run log

closes #11228
PR #11234 (2022-07-28): feat(app): support module JSON v6 commands in run log
closes #11228

# Overview

This adds module (all 4 modules) JSON v6 command support to run log as well as `waitForDuration`

# Changelog

- add a bunch of `case`s to `StepText` and add test coverage for each case
- add `HeaterShakerDeactivateShakerRunTimeCommand` to a type of `ModuleRunTimeCommand`

# Review requests

- upload a v6 command protocol with modules (you can do this by running PD with the H-S ff turned on), look through the run log, the text should display the command text correctly instead of only displaying the `commandType`
- [... cut]
commit 5f0cc21a (2022-07-28): feat(app): support module JSON v6 commands in run log (#11234)

closes #11228
closed 2022-07-28 by jerader