Opentrons/opentrons #11263: bug: App window goes blank when I navigate to the protocol tab

moutonzr, 2022-07-29:
### Overview

I can connect robot to computer, but when I navigate to load the protocol using the app, or if I directly go into the run tab, the window goes blank, and I have to exit the app and reopen it to get any functionality.

### Operating system

Windows

vabruzzo, 2022-08-02:
Hi @moutonzr , thank you for reporting this bug. Can you share what OS you are on, what version of the app you are using, and the protocol you are using, if possible. Thank you.

koji, 2023-04-15:
We will close this since We haven't heard any response for a while.
If still this happens to you, please re-open the issue with additional information.

closed 2023-04-15 by koji