Opentrons/opentrons #11279: App [Housekeeping]: Replace old font size with TYPOGRAPHY const

koji, 2022-08-02:
There are some components using `FONT_SIZE_` from `@opentrons/components` instead of using `TYPOGRAPHY.fontSize`.

For example,

`app/src/organisms/Devices/ProtocolRun/LabwareInfoOverlay.tsx` L53

```ts
<Box
  backgroundColor={hover ? COLORS.blue : OVERLAY_BLACK_80}
  borderRadius="0 0 0.4rem 0.4rem"
  fontSize={FONT_SIZE_CAPTION}. <-- L53
  padding={SPACING.spacing2}
  color={COLORS.white}
  id={`LabwareInfoOverlay_slot_${labwareId}_offsetBox`}
>
```

L53 should be `fontSize={TYPOGRAPHY.fontSizeCaption}`

font size const is in https://github.com/Opentrons/opentrons/blob/edge/components/src/ui-style-constants/typography.ts

If you use VSCode, you can check the value easily like below
[image]

**The target components for this should be components that use a component/components from `atom` or `molecules`**

MichaelMulieri, 2022-08-04:
There are variables that don't have mapping to an equivalent size in TYPOGRAPHY.  
{FONT_SIZE_BODY_1}
{FONT_SIZE_HEADER}
{FONT_SIZE_DEFAULT}
{FONT_SIZE_INHERIT}
Does that seem right or am I overlooking something?

vabruzzo, 2022-08-04:
Thanks for your contribution! I think we can add the following to `components/src/ui-style-constants/typography.ts`

```
export const fontSizeHeader = '1.125rem' // 20px
export const fontSizeDefault = '1rem' //  16px
export const fontSizeBody = '0.75rem' // 12px
export const fontSizeInherit = 'inherit'
```

With that said, we might want to use `fontSizeH1` for `FONT_SIZE_HEADER`, and omit `fontSizeDefault` entirely and just remove the property since I think text will default to that size on its own. We might also be able to use `fontSizeP` in place of `FONT_SIZE_BODY`.

@koji do have any guidance on these four?

koji, 2022-08-04:
@MichaelMulieri @vabruzzo 
The sizes in `TYPOGRAPHY` have been finalized by Mel (UI/UX designer).
I think components that use those 4 font sizes will be replaced with the new design. So we can leave them.

koji, 2022-08-04:
@MichaelMulieri @vabruzzo 
I just checked 4 font sizes. They are used by components that will be updated by future releases.
So we can leave them.

MichaelMulieri, 2022-08-04:
@koji @vabruzzo 
Ok sounds good.  Thank you!

closed 2022-08-08 by vabruzzo