Opentrons/opentrons #11281: bug: 6.0.0 App: Changes to advanced pipette settings do not take effect

braldrich, 2022-08-02:
### Overview

Changes to values in the advanced pipette settings menu don't take effect and there's no response from the app when clicking the "Confirm" button. 

Tested with all the values that can be edited in the pipette settings. Not seeing any response from the "Reset all" button either. 

### Steps to reproduce

1. Open App 
2. Devices Tab
3. Select connected Robot 
4. Open Pipette Settings Menu from one of the pipette tiles 
5. Change any of the values, hit the enter key on keyboard or click out of the dialog box
6. Click confirm 
7. Close out Pipette settings 
8. Re open settings menu 

### Current behavior

No response from the confirm button press, the values do not change after editing them. No values change, and it doesn't work on either pipette. 

[combined.log](https://github.com/Opentrons/opentrons/files/9245732/combined.log)

https://user-images.githubusercontent.com/95253244/182475763-44a423d2-6079-43b9-b017-0eddc7fc148e.mov

 

### Expected behavior

The value should hold when you edit it, and there should be a response from the confirm button. 

### Operating system

Mac

### System and robot setup or anything else?

6.0.0 
Connected via ethernet (OT-2R)

braldrich, 2022-08-02:
[:data:pipettes_files.zip](https://github.com/Opentrons/opentrons/files/9245956/data.pipettes_files.zip)
[pipette_settings_confirmFail.har.zip](https://github.com/Opentrons/opentrons/files/9245964/pipette_settings_confirmFail.har.zip)

vabruzzo, 2022-08-04:
Thank you for reporting this bug @braldrich, we appreciate the support. We are working on a bugfix for this now, we are on track to release a hotfix for the 6.0 release in the next few days and it will include this fix.

mcous, 2022-08-12:
Fixed in v6.0.1

PR #11288 (2022-08-05): fix(app): connect pipette settings form's submit button
# Overview
Closes #11281.
When users edit pipette settings and click submit, the submit button does not handle the submit and no settings are saved. To fix this we added an `id` to the form, and a `form` prop that references the form's `id`. This fixes the submit functionality.

We had a remaining data de-sync issue: we were using redux via `updatePipetteSettings` to mutate the pipette data and `usePipetteSettingsQuery` to retrieve the data. This was causing some data de-syncing. This fixes data sync issues by using a redux selector to re [... cut]
closed 2022-08-12 by mcous