Opentrons/opentrons #11338: feat: Remember the preferred way of sorting protocols

VasBrd, 2022-08-12:
### Overview

In the new 6.0.1 app there is a Protocols page that lists uploaded protocols, but they're sorted alphabetically by default. Request to make it remember last used way of sorting.

### Implementation details

Our usecase involves making a new protocol for every run (from a template), making it much more useful to sort by "Recent" instead of "Alphabetical". With every app launch it defaults to the latter, making one go through selecting "Recent" everytime and potentially creating confusion when one forgets to do that. 

### Design

Have the app use whatever sorting method was selected previously.

bndo, 2022-09-20:
Thanks for submitting this as well! Great feedback that we'll look to incorporate in future iterations.

koji, 2022-12-03:
The PR for this has been merged into `edge`.
https://github.com/Opentrons/opentrons/pull/11787

This will be available in 2023's first release.

closed 2022-12-03 by koji