Opentrons/opentrons #11459: bug:

VasBrd, 2022-09-13:
### Overview

In the new 6.0.1 app the "Lights" slider under Devices -> *OT2* -> Controls takes up to about 10 seconds to respond. This was not the case in the previous version, where the lights toggled on/off instanteniously.

### Current behavior

 "Lights" slider takes up to about 10 seconds to respond

### Expected behavior

The lights turn on/off instantly when the slider is toggled.

### Operating system

Windows

### System and robot setup or anything else?

App Software Version: 6.0.1
Connection: USB

bndo, 2022-09-16:
Thanks for submitting this! We'll take a look and report back.

VasBrd, 2022-10-13:
Hello @koji 
Thanks for looking into this issue!

1. When I say previous version I mean before the major update, it was 5.x.x (don't remember the exact value and bot sure where can one find it)

2. The issue does indeed happen when reloading the app and connecting to the robot for the first time. It does not happen if you switch panels (i.e. to "protocols" or to another robot) and then return back, but it does happen after prolonged idling time even if the app remains open, including those times when a protocol is running (which usually takes 20~60 minutes, with no additional human input while it runs)

koji, 2022-10-13:
@VasBrd Thank you!

closed 2023-05-11 by koji