Opentrons/opentrons #11471: Linux Ubuntu 22.04 - App is unable to open

chewyalvin, 2022-09-16:
Dear all, I am facing issues opening and using the opentron app on my linux machine (Ubuntu 22.04 LTS) the icon does not work "Opentrons.AppImage", opening is via cmd line ./Opentrons.AppImage results in the error messages below, opening via sudo --no-sandbox mode manages to open the GUI but with error messages too. I have also attached the error logs, hopefully someone can advice on what to do please.
xx@xx:~/Desktop$ ./Opentrons.AppImage 
13:31:56.888 [log] info: Level "error" and higher logging to /home/dlsw3/.config/Opentrons/logs/error.log
13:31:56.908 [log] info: Level "debug" and higher logging to /home/dlsw3/.config/Opentrons/logs/combined.log
13:31:56.909 [log] info: Level "info" and higher logging to console
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(Use `opentrons --trace-warnings ...` to show where the warning was created)
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to
[combined.log](https://github.com/Opentrons/opentrons/files/9580992/combined.log)
 [Console]. Use emitter.setMaxListeners(
[error.log](https://github.com/Opentrons/opentrons/files/9580993/error.log)
) to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:43690) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
13:31:58.363 [main] info: Starting App
13:31:58.746 [ui] info: Loading file:///tmp/.mount_OpentrEwqiI7/resources/app.asar/ui/index.html
13:31:58.790 [main] info: Creating renderer logger
13:31:58.866 [protocol-analysis/getPythonPath] info: Python environment selected { pythonPath: '/tmp/.mount_OpentrEwqiI7/resources/python/bin/python3' }
(node:43690) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[43690:0916/133202.628311:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)

koji, 2022-09-16:
@chewyalvin 
Could you try [this](https://itsfoss.com/cant-run-appimage-ubuntu/)?

chewyalvin, 2022-09-19:
> @chewyalvin Could you try [this](https://itsfoss.com/cant-run-appimage-ubuntu/)? 6.0.0 & 6.0.1 aren't built for Linux Ubuntu 22.04.

The fixed recommended does not fix the issue. Any other suggestions?

chewyalvin, 2022-09-19:
> @chewyalvin 
> your machine's architecture is `x86_64`, isn't it?

Yes it is

chewyalvin, 2022-09-26:
Yes I have done chmod to all permissions:
[image]

I have also used the GUI to enable the execute permission "Allow executing file as a program"
[image]

double-click (+ update permission) does not work, nothing happens upon double-click.

uname -a
Linux dlsw3 5.15.0-47-generic #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

koji, 2022-09-26:
@chewyalvin thank you for sharing the information on your environment.
The screenshots you posted looks okay.

Seems that the issue came from low level stuff.
Unfortunately, it is very hard to solve this issue since we could not reproduce the same issue on our end.

There are 2 suggestions for you.
1. set up dev env and build the app from source code ([this v6.0.1 branch](https://github.com/Opentrons/opentrons/tree/v6.0.1))
2. wait for the v6.1.0 (the version supports Ubuntu 22.04)

If you need help for 1, feel free to send a mention to me.

Sorry for inconvenience to you.

koji, 2022-09-26:
@chewyalvin let me check the schedule.

in terms of, dev env set up you can see [this](https://github.com/Opentrons/opentrons/blob/edge/DEV_SETUP.md).
If you have any issues on dev env, feel free to send a mention to me. I can help.

koji, 2022-09-26:
@chewyalvin probably v6.1 will be out in a couple of days if there is no issue.

Javier1704, 2022-09-29:
@koji It seems that the v6.1 update is out already, but I was only able to download it from an existing ver of the app (Windows OS). download link provided on the Opentrons site is still for the v6.0.1 ver of the app. Do you have an installation guide for the latest ver of the app for Linux Ubuntu 22.04 OS?

koji, 2022-09-29:
Hi, @Javier1704 we will update the download link soon.

koji, 2022-09-30:
@chewyalvin @Javier1704 
The download page has been updated.
https://opentrons.com/ot-app/

koji, 2022-10-01:
We haven't heard about the issue since the release. We close this issue.
If you face the same issue, please reopen this.

Javier1704, 2022-10-03:
@koji The response time of the GUI interface is ~10 mins when I open from the app icon and it takes ~5 minutes for the connection to the OT-2, 
[error.log](https://github.com/Opentrons/opentrons/files/9694763/error.log)
[combined.log](https://github.com/Opentrons/opentrons/files/9694764/combined.log)

koji, 2022-10-05:
Hi, @Javier1704 could you share your machine information with me?

- your cpu information
- output of `uname -a`

Javier1704, 2022-10-05:
@koji 
Linux dlsw3 5.15.0-48-generic #54-Ubuntu SMP Fri Aug 26 13:26:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

koji, 2022-10-05:
@Javier1704, could you also tell me your machine's cpu model?

Javier1704, 2022-10-06:
@koji 
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         46 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  12
  On-line CPU(s) list:   0-11
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Xeon(R) CPU E5-2620 0 @ 2.00GHz
    CPU family:          6
    Model:               45
    Thread(s) per core:  2
    Core(s) per socket:  6
    Socket(s):           1
    Stepping:            7
    CPU max MHz:         2500.0000
    CPU min MHz:         1200.0000
    BogoMIPS:            3990.63

yfujita-skgcat, 2022-10-06:
What happens if you add "--disable-gpu-sandbox"?
> Opentrons-v6.0.1-linux-b21052.AppImage --disable-gpu-sandbox

koji, 2022-10-06:
@yfujita-skgcat thank you for your suggestion.

@Javier1704, could you try what yfujita-skgcat suggested? 
Unfortunately the current app isn't optimized for xeon.

closed 2022-10-08 by koji