Opentrons/opentrons #11484: bug: Missing deck_calibration.json and robot_settings.json

aperkins19, 2022-09-20:
### Overview

Hey Devs :)

The robot is not remembering "labware offset" settings.

When I calibrate it through the GUI (i.e. import a protocol file, adjust the calibration to the specific labware, "apply labware offsets" and then run the protocol) it works fine and visibly, the robot is clearly remembering those settings - both deck calibration and specific labware.

However, 

Due to our protocols being highly dynamic, we run scripts with accompanying files ( .py & .json) that need to be read in. We transfer them all into a subdirectory in */user_storage* and `opentrons_execute`.

This works fine except that doing so yields the following warning before the protocol runs:

[image]

As you can see, the system can't see the required: `deck_calibration.json` and `robot_settings.json`.

Lets look at the contents of *\data*:

[image]

Presumably these files are written every time you do a deck cali or labware offset adjust.

I'm not sure why these files are A) not being written or B) not persisting.

Please help :) 

*p.s. can this in Advanced Settings help?*

[image]

## Extra info

*Connecting to OT2 via USB*

OT2 info:

[image]

The directory with the files:

[image]

### Expected behavior

It would be really useful for us to able to perform a deck cali and protocol specific labware offset from the terminal.

### Operating system

Windows

aperkins19, 2022-09-20:
I tried stopping the opentrons-robot-server:

[image]

But it didn't help :')

mcous, 2022-09-20:
@aperkins19 these log warnings are from old settings files that are no longer used and can be ignored. Labware offsets are not persisted outside the scope of an individual run as of `v5`.

> *p.s. can this in Advanced Settings help?*
> 
> [image]

Yes, since you are using `opentrons_execute`, you should be using this setting along with the [`labware.set_offset` API](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.labware.Labware.set_offset). Using `opentrons_execute`, you _must_ use `labware.set_offset` in your protocol to use labware offsets.

Please see [this support article](https://support.opentrons.com/s/article/How-to-apply-labware-offset-values-in-Juypter-Notebook-and-opentrons-execute) for more details

closed 2022-09-20 by mcous

aperkins19, 2022-09-26:
@mcous 

Thanks for getting back to me.

All the best.