Opentrons/opentrons #11539: bug: Use the version 6.1 with WIN7

prscustodio, 2022-10-06:
### Overview

HI!

I need use the version 6.1 in win 7, but my protocols does't analyzes. I'm thing which this occur because the Python installed together opentrons software not is compatible with a WIN 7, can you help me to use this version in this version of windows?

### Operating system

Windows

ethanfjones, 2022-10-10:
Hello @prscustodio - to ensure you get a quick response in the future you can always contact our support team at support@opentrons.com.

Our app does not support Windows 7 and we cannot guarantee that our app will function properly on that OS. We have actually seen that Opentrons App version 6+ cannot properly analyze protocols. The workaround would be to [downgrade](https://support.opentrons.com/s/article/Downgrade-to-an-older-software-version) to an app version that has worked for you in the past or upgrade your operating system to one that our app [supports](https://support.opentrons.com/s/article/Get-started-Supported-operating-systems-for-the-Opentrons-App).

mcous, 2022-10-12:
As Ethan said, since Microsoft no longer supports Windows 7, we cannot guarantee that the app will work either.

- If you are able to set up a working Python environment on your computer with the `opentrons` package installed, you can use the app's advanced settings to configure the app to use your own Python environment instead of its built in one
- Even if the app is unable to analyze the protocol, assuming the app is working at all, you should be able to send and run your protocol to your OT-2 regardless

The support team may be able to provide further guidance on one or both of these suggestions, but again, we do not support Windows 7, so suggestions are the best we'll be able to do here.

closed 2022-10-12 by mcous