Opentrons/opentrons #11542: feat:

naikymen, 2022-10-07:
### Overview

Hi!

First of all I wanted to say thanks for this great project.

Our request is about making software for pipetting robot compatible / interoperable. Specifically, we would like to add compatibility with the hardware/electronics of other CNC machines (for example, to reuse a 3D printer / CNC mill).

We'd like to invite you to chat for a bit, and decide which is the happiest place to add interoperability between different machines.

### Implementation details

There might be a sweet-spot for adding interoperability to open-hardware automation.

The high level JSON protocol produced by the protocol designer seems nice, because it seems rely little on the specific CNC hardware in the machine (other than knowing the available tools). It is also easier to understand by a human.

The python API also looks promising. But I would like to know how tied it is to the "drivers" for the CNC hardware in the OT2.

### Design

Here is my understanding of pipetting machines at birds eye view:

[image]

The top one is my impression of the OT, and the bottom one is the one we're developing.

> A note on motivation: there are a couple OT2's at our faculty, but we had a few incentives to start from scratch: (0) they are not easy to get to, and purchasing an OT2 or it's parts is out of our budget, (1) to have fun making stuff, and (2) because we had a few ideas to improve the machine (which we'd be happy to share).

### Acceptance criteria

- [ ] Have a chat.
- [ ] Define possible interfaces for interoperability.
- [ ] Decide which one is the happiest.
- [ ] Collaborate on its development.

Thanks again :)

mcous, 2022-10-12:
Hey @naikymen, it's very cool to hear that y'all are also working on an open-source liquid handler! I'm sure we can find someone(s) on our team to have a chat about our software architecture and how it relates to our hardware interactions.

To be pragmatic, there's not a high likelihood of us integrating third party hardware support into the core Opentrons software. This is purely a practical, not philosophical! We simply have our company goals / priorities and a limited number of engineers to accomplish them with.

That being said, obviously Opentrons/opentrons is an open-source project, and there are various layers of abstraction that could be re-used or re-implemented by your project. Immediately, maybe there's an opportunity for y'all to fork or depend on this repository. Looking down the road, there's been long-standing interest in implementing a plugin system of sorts into the Opentrons core, which could be another avenue for integration.

I'll see what architectural documentation I can dig up and/or generate over the coming weeks. In the mean time, do you have an email address that would be a good place to continue this conversation?

naikymen, 2022-10-12:
> do you have an email address that would be a good place to continue this conversation?

Of course! it's nmendez@iib.unsam.edu.ar

> there's not a high likelihood of us integrating third party hardware support into the core Opentrons software.

That's alright, asking your company to write drivers for a different machine sounds like a bit much :P

We were mostly hoping to find a more abstract level of interaction between different machines/software. Perhaps at the protocol level, but surely before the drivers.

I believe that the challenge will be around making a single GUI or API for machines with different (hardware) capabilities.

> there's been long-standing interest in implementing a plugin system of sorts into the Opentrons core

That would be really great. I imagine that at least some of the features we've needed could be added as plugins.

Thank you for your efforts @mcous, I hope to hear from you soon!

naikymen, 2022-12-19:
Hello!

We've not been contacted yet, and I wanted to check if there are any updates on this from OT's side.

In the meantime, our efforts have moved to a different approach:

1. Polishing our GUI,
2. moving our robot to Klipper[^1] firmware,
3. using PyLabRobot's API[^2],
4. writing a translator for OT's protocols (in either JSON or API format) to PyLabRobot.

Everyone is invited to join this open interops party :)

Cheers!

[^1]: [Klipper](https://www.klipper3d.org/) is a 3D-printer/CNC firmware more _agnostic_ to micro controllers.
[^2]: [PyLabRobot](https://forums.pylabrobot.org/) is meant to be robot _agnostic_; we'll write a "generic" backend for our bot.