Opentrons/opentrons #11570: Canceling a protocol outside the app

caroline-wyzer, 2022-10-13:
A few months ago, we had a pipette crash into the tip box following our protocol cancellation using a try/catch loop. It seemed like this occurred because of the interpreter, which would stop the current step, run a recover and reset block, and then finish the step it was on before, probably without certain variables such as labware calibration, etc, resulting in our crash.

Has this issue been resolved? And if not, is there any way we can have our code get around it (since the app allows cancellation, is there some code that allows this to be done safely)? We would ideally like to be able to cancel protocols, because right now if we make any error in the setup, we have to wait an hour before we can use that robot again.

koji, 2022-10-15:
Hi @caroline-wyzer,

The app uses [this function](https://github.com/Opentrons/opentrons/blob/edge/api-client/src/runs/createRunAction.ts) via react-query to cancel a protocol run. 
The readme in api-client says that `npm install` can install `api-client` as a package, but we haven't published it as a package so the readme doesn't work. (sorry for the inconvenience to you).
If you plan to use js/ts, you will need to clone this repo and build api-client and install it as a package with npm/yarn. Our repo is using `yarn` so I highly recommend you use yarn.

If you plan to use python, probably [this new python](https://github.com/rickwierenga/opentrons-python-api) wrapper will be good.

The support team may be able to provide further guidance on this. [support@opentrons.com](mailto:support@opentrons.com). When you reach out to the team, please provide your environment info(app version and robot-server version) with them. Then they can response to you quickly.

mcous, 2022-10-31:
Hey @caroline-wyzer, I'm going to close this in favor of the original ticket #11360. There have been no changes to SSH based execution, but I'm going to sound off in that thread with an alternative solution

closed 2022-10-31 by mcous