Opentrons/opentrons #11586: feat: Variable metaData

jjmc0000, 2022-10-18:
### Overview

Problem: Due to the nature of the 6.1 update, my users update their protocols to their opentrons units and the metadata is read and made the title of their protocol.  Oftentimes my users will change just a small thing in the python file, such as a global variable name.  This becomes rather cumbersome and confusing, as when my users re-upload this protocol for pre-simulation, the name of the protocol remains the same as the previously uploaded one, causing a lot of fussing about and deleting old protocols constantly.  

Solution: It would be really nice if we could make our metadata variable.  Example where my user is changing the number of plates they want to run:

NUM_PLATES = 8

It would be nice if my metadata could be:

metadata = {protocolName: ("Plate Coating {} plates").format(NUM_PLATES)}, "description": "Coats Plates!",
    "author": "Joe Caponi <joe@mosameat.com>",
    "apiLevel": "2.10",
}

Then my name would change based upon how many plates the user is running (or whatever variable they need to change!) 

### Design

[image]

### Acceptance criteria

1. Upload protocol to opentrons 6.1 app
2. See name
3. Change variable in python file
4. Upload protocol to same opentrons 6.1 app
5. See if protocol name has changed 
6. If protocol_name_change == True: 
7. Celebrate!

SyntaxColoring, 2022-10-20:
Thanks for the detailed feature request!

I think this is something that we can theoretically do, if we take some care. Some background:

Currently, we parse `metadata` statically, without actually importing and executing the Python file. We do this for two reasons:

* Responsiveness.

  The design of the Opentrons App assumes that these basic protocol details—name, author, etc.—will be available immediately after upload. Which I think is reasonable.

  But a Python variable can take arbitrarily long to compute. Imagine if `NUM_PLATES` in your example were the output of some crazy long complicated algorithm. Or if it had an infinite loop bug.

* Safety.
  
   It lets us avoid importing and executing the Python file until after we've read `apiLevel` and determined, with some confidence, that we're looking at a Python Protocol API v2 protocol. This stops certain mistakes from causing unexpected runtime behavior. In particular, old Python Protocol API *v1* protocols [controlled the robot with top-level statements](https://docs.opentrons.com/v1/index.html#how-it-looks), uncontained in any `run()` function. We no longer support APIv1, but if someone uploaded one of those files, it still might cause unexpected things to happen immediately upon upload if we naively imported it.

But a cost of this staticness, as you've noted, is that `metadata` can only contain simple things like string literals, not "complex" things like variables.

-----

We could solve the responsiveness problem with some kind of reasonable timeout, like 5 seconds. And we could solve the safety problem by continuing to read `apiLevel` statically, but switching to actual Python execution for reading everything else.

jjmc0000, 2022-10-21:
Hi Max!

Thanks for looking at my suggestion! Honestly, anything that would allow me
to dynamically change the name that shows up in the protocols page would be
really helpful, not only for me but likely for other users too considering
how the app has changed.

I think the api level being read statically is a great solution combined
with a timeout. Let me know if you decide to implement this and thanks
again for looking into it and taking me seriously!

Cheers

Joe

-- 

Joseph Caponi
Laboratory Robotics Technician
–
Watermolen 28
6229 PM Maastricht
The Netherlands

[+31 684034016]
mosameat.com
<https://mosameat.com/>
***@***.*** <https://twitter.com/mosa_meat>.com

<https://twitter.com/mosa_meat>