Opentrons/opentrons #11780: bug: crash on startup on linux. GPU related

mahogny, 2022-11-25:
### Overview

downloaded the app, run in terminal, and I get below. why is it even trying to access my GPU? that is the origin of the crash

3:17:11.327 [log] info: Level "error" and higher logging to /home/mahogny/.config/Opentrons/logs/error.log
13:17:11.342 [log] info: Level "debug" and higher logging to /home/mahogny/.config/Opentrons/logs/combined.log
13:17:11.343 [log] info: Level "info" and higher logging to console
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(Use `opentrons --trace-warnings ...` to show where the warning was created)
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
(node:736717) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [Console]. Use emitter.setMaxListeners() to increase limit
13:17:12.046 [main] info: Starting App
13:17:12.266 [ui] info: Loading file:///tmp/.mount_OpentrU9fZd0/resources/app.asar/ui/index.html
13:17:12.280 [main] info: Creating renderer logger
13:17:12.302 [protocol-analysis/getPythonPath] info: Python environment selected { pythonPath: '/tmp/.mount_OpentrU9fZd0/resources/python/bin/python3' }
(node:736717) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[736717:1125/131713.310455:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)

### Steps to reproduce

./Opentrons-v6.1.0-linux-b22274.AppImage 

### Operating system

Linux

### System and robot setup or anything else?

Opentrons-v6.1.0-linux-b22274.AppImage 

not connected at all

(base) mahogny@noveros:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

koji, 2022-11-25:
@mahogny 
Can you try the following?

```zsh
Opentrons-v6.1.0-linux-b22274.AppImage --disable-gpu-sandbox
```

koji, 2022-11-28:
@mahogny 
If the above comment doesn't work for you, please reach out to `support team`([support@opentrons.com](mailto:support@opentrons.com)).

Thank you!

koji, 2022-12-02:
Please reopen this ticket if you need.

closed 2022-12-02 by koji