Opentrons/opentrons #11803: bug:

Maggie-Lam, 2022-12-01:
### Overview

No robot found with USB or wifi connection. This started on 2022-11-29, after working fine for months. The app sees the USB connection, but cannot connect to the robot. Issue was reproduced with versions 5 and 6 on two different computers. 

We did the connection troubleshooting, but are still waiting to hear from support. 

The ethernet to USB adapter is working, the wifi on the Raspberry PI is working, and all the status lights are good on the RPI and the motor controller board. 

We also disconnected and reconnected all the cables between the RPI and board, as well as trying to directly connect the RPI to a router via ethernet, and directly to the computer through a different ethernet adapter by entering the IP manually. 

We can see the RPI wifi on our router when it's on, but it won't connect to the app. 

### Steps to reproduce

Unsure if reproducible, may be our hardware. Issue persists between versions 5 and 6 of the app on two different computers. 

### Current behavior

Software opens correctly, but cannot see or connect to robot when robot is on and the status light is solid. 

### Expected behavior

App should be able to recognize and connect to the robot when there is no apparent robot issue. 

### Operating system

Mac

### System and robot setup or anything else?

No robot found with both app version 5 and 6. Tried both USB and WIFI.

koji, 2022-12-01:
@Maggie-Lam 
In terms of v6, are you using v6.0.0 or v6.1.0?
Also when you open `Devices` on v6, does the app show No found screen or Available and Unavailable list? 

Have you tried to add an ip address from App Settings?
Gear icon -> `Set up connection` button

Maggie-Lam, 2022-12-01:
@koji
Thanks for the quick response!

We are using v6.1.0. The app shows the "No robots found" screen. 

Yes, we have tried adding an ip address.

koji, 2022-12-01:
@Maggie-Lam 
Thank you for your response.

>Yes, we have tried adding an ip address.
When you tried that, the app couldn't detect the ip address, could it?

On your end, is there any network update?
If there is no update on network, could you try to reinstall the Desktop app, please? (if you haven't tried that.)
In addition could you check that your Mac and OT-2s are on the same network?

koji, 2022-12-05:
@Maggie-Lam 
Are you still facing this issue?

SPC-Archer, 2022-12-08:
Just wanted to add that we have faced the same problem using a Mac (OSX 12.6) connected via USB. 

Occasionally our OT-2 cannot be found by the app (in both v6.1.0 and v6.2.0). Unfortunately, we havent noticed a pattern yet.

The robot is at the correct IP address (confirmed by using `arp -a` in Terminal) but is just not found by the app. The robot disappears from the list when unplugged and re-appears when plugged back in but still isn't found by the software. 

Restarting the robot and software doesn't do anything but restarting the computer seems to work.

Maggie-Lam, 2022-12-08:
@koji
Sorry for not replying sooner. We were actually able to resolve the issue by [reflashing the SD card](https://support.opentrons.com/s/article/Reflashing-the-OT-2-s-SD-card) as suggested by the support team. Thank you very much again for helping us troubleshoot.

koji, 2022-12-08:
@Maggie-Lam thank you for your response. Glad to hear that!

koji, 2022-12-08:
Hi @SPC-Archer

closed 2022-12-08 by koji

SPC-Archer, 2023-01-05:
Hi @koji, 

The connection issue has appeared again. This time it appeared after coming back after our Christmas break where the OT2 was turned off (at the switch near the network/power ports) and the connected laptop was asleep for about 10 days. 

Running `arp -a` to list the connected devices shows the OT2 and I believe the OT2 is identifying itself as it appears as `ot2cep20220720r01.local` on `en0`. 

However, the OT2 cannot be communicated with: the HTTP request you suggested to reset settings times out as does an SSH request. 

Given the OT2 was off and the laptop was on, I figured its more likely the fault the MacBook than the OT2 so I tried resetting the network adapters and I think I found the issue.  

Switching off wifi adapter (`sudo ifconfig en0 down`) seemed to force the connection to the other ethernet adapter (`en4`) and then suddenly the OT2 is discoverable in the app and I can SSH into it. Once I turned the wifi adapter on again the OT2 stayed on `en4`.

Not sure why MacOS was attempting to connect to it via the wifi adapter - wifi is switched off on the robot as it cannot connect to the enterprise wifi we have here. My guess is that there is a bug somewhere in MacOS.

Probably not much you can do on your end to stop this happening in the future, but maybe something as simple as turning the wifi off from the task bar might fix it (I will try it and report back next time this happens).

koji, 2023-01-05:
Hi @SPC-Archer 
Thank you for the details and really appreciate your help!
Our team keeps an eye on this issue.

Thank you again!