Opentrons/opentrons #11899: feat: Externally loaded components (PD)

protdesigner, 2022-12-16:
### Overview

Hello,
I have a small UI components library (eg: on bit.dev) with some custom UI elements,
can these be loaded on the current protocol-designer from external resources and utilized by the current PD webpack instance that you have?

shlokamin, 2023-01-13:
Hey there! yes — we are an open source company so you're free to fork our monorepo and change the internals of PD however you'd like. 

You would just need to:

1. Fork the monorepo
2. Make sure you follow our dev setup instructions for the [monorepo](https://github.com/Opentrons/opentrons/blob/edge/DEV_SETUP.md) and [PD](https://github.com/Opentrons/opentrons/tree/edge/protocol-designer)
3. Add the components library as a dependency in [PD’s package.json](https://github.com/Opentrons/opentrons/blob/edge/protocol-designer/package.json)
4. Run `make setup-js` (or `yarn` ) from the root of the monorepo
5. Use the components library like any other dependency in PD's source code!

closed 2023-01-13 by shlokamin

protdesigner, 2023-01-16:
OK Amin, thanks for the update. To complete this issue from my end I will also need to know if I can call an external component using federated remote loading? I think the main server is missing some relevant dependencies to support that functionality.

Thanks and all the best.

braldrich, 2023-01-18:
@shlokamin Thanks for working with @protdesigner ! Do you have any insight into their second question?

shlokamin, 2023-01-18:
Hey @protdesigner!

> I think the main server is missing some relevant dependencies to support that functionality.

PD is an entirely client side application, there are no servers. 

Regarding module federation, PD is not a microfrontend application. It is a single application bundled by webpack. 

If you would like to pull PD into another system and run it as part of a microfrontend system, you are welcome to do so! We expose PD's webpack config and build internals in our monorepo.

protdesigner, 2023-01-19:
> Hey @protdesigner!
>
>
> PD is an entirely client side application, there are no servers.
>
> Regarding module federation, PD is not a microfrontend application. It is a single application bundled by [...]

Thanks for the update  on this matter, I already checked the dependencies and relevant .config files and determined that the current webapp version does not support the federated microfront-end features (as stated above). I will try and see what can be done in either case, to have the external loading behaviour. Thanks for this input from your end, this has been helpful to proceed towards the correct direction.