Opentrons/opentrons #11900: bug: Cannot upload protocols to Opentrons 6.2.0

kabalak, 2022-12-16:
### Overview

After updating to the Opentrons App 6.2.0 and the Server App 6.2.0 we cannot upload protocols reliably to our Opentrons; regardless whether it is via WiFi or Ethernet cable, we get the same error message:

**opentrons.protocol_reader.protocol_reader.ProtocolFilesInvalidError: 20221215 single tubule_protein extraction.json is not valid JSON.**

We have produced the JSON file with the current version of the ProtocolDesigner on the Opentrons website and the JSON Lint checkers also state, that the file contains valid JSON.

### Steps to reproduce

[serial.log](https://github.com/Opentrons/opentrons/files/10245257/serial.log)
[api.log](https://github.com/Opentrons/opentrons/files/10245258/api.log)
[server.log](https://github.com/Opentrons/opentrons/files/1024526
[protein_extraction_Transfer.zip](https://github.com/Opentrons/opentrons/files/10245262/protein_extraction_Transfer.zip)
0/server.log)

### Current behavior

Currently, we cannot use any protocols on the Opentrons, due to the mentioned error, that our JSON file is not correct. And it seems an error within the Opentrons platform.

`Dec 16 10:26:28 OT2CEP20220107R05 opentrons-api[204]: Instruments found: {<Mount.LEFT: 1>: <Pipette: P300 8-Channel GEN2 1916931344>, <Mount.RIGHT: 2>: <Pipette: P20 8-Channel GEN2 1916930896>}
Dec 16 10:26:40 OT2CEP20220107R05 opentrons-api[204]: BEGIN (implicit)
Dec 16 10:26:40 OT2CEP20220107R05 opentrons-api[204]: SELECT protocol.id, protocol.created_at, protocol.protocol_key 
                                                      FROM protocol
Dec 16 10:26:40 OT2CEP20220107R05 opentrons-api[204]: [generated in 0.00211s] ()
Dec 16 10:26:40 OT2CEP20220107R05 opentrons-api[204]: COMMIT
Dec 16 10:27:16 OT2CEP20220107R05 opentrons-api[204]: Error response: 500 - LegacyError - 
Dec 16 10:27:16 OT2CEP20220107R05 opentrons-api[204]: Exception in ASGI application
                                                      Traceback (most recent call last):
                                                        File "usr/lib/python3.7/site-packages/uvicorn/protocols/http/httptools_impl.py", line 371, in run_asgi
                                                        File "usr/lib/python3.7/site-packages/uvicorn/middleware/proxy_headers.py", line 59, in __call__
                                                        File "usr/lib/python3.7/site-packages/fastapi/applications.py", line 208, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/applications.py", line 112, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/middleware/errors.py", line 181, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/middleware/errors.py", line 159, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/middleware/cors.py", line 78, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/exceptions.py", line 82, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/exceptions.py", line 71, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/routing.py", line 580, in __call__
                                                        File "usr/lib/python3.7/site-packages/starlette/routing.py", line 241, in handle
                                                        File "usr/lib/python3.7/site-packages/starlette/routing.py", line 52, in app
                                                        File "usr/lib/python3.7/site-packages/fastapi/routing.py", line 220, in app
                                                        File "usr/lib/python3.7/site-packages/fastapi/dependencies/utils.py", line 550, in solve_dependencies
                                                        File "usr/lib/python3.7/site-packages/robot_server/protocols/dependencies.py", line 69, in get_protocol_store
                                                        File "usr/lib/python3.7/site-packages/robot_server/protocols/protocol_store.py", line 150, in rehydrate
                                                        File "usr/lib/python3.7/site-packages/robot_server/protocols/protocol_store.py", line 420, in _compute_protocol_sources
                                                        File "usr/lib/python3.7/site-packages/anyio/_backends/_asyncio.py", line 564, in __aexit__
                                                        File "usr/lib/python3.7/site-packages/anyio/_backends/_asyncio.py", line 601, in _run_wrapped_task
                                                        File "usr/lib/python3.7/site-packages/robot_server/protocols/protocol_store.py", line 410, in compute_source
                                                        File "usr/lib/python3.7/site-packages/opentrons/protocol_reader/protocol_reader.py", line 105, in read_saved
                                                      opentrons.protocol_reader.protocol_reader.ProtocolFilesInvalidError: 20221215 single tubule_protein extraction.json is not valid JSON.`

### Expected behavior

With prior versions, we could do this without any trouble or problems.

### Operating system

Windows

### System and robot setup or anything else?

Opentrons App 6.2.0 and server version 6.2.0, freshly installed. Robot is connected to via WiFi normally, but we also tried Network connection.

koji, 2022-12-19:
Hi @kabalak 
Could you try `Clear Runs History`?

https://support.opentrons.com/s/article/Factory-resetting-the-OT-2

koji, 2022-12-21:
If you need to keep this open, please re-open.

closed 2022-12-21 by koji