Opentrons/opentrons #12006: bug: python3.11 coroutine dependency broke

Koeng101, 2023-01-14:
### Overview

`opentrons_simulate` on python3.11 does not work due to coroutines being updated.

### Steps to reproduce

- Install python3.11
- Install opentrons (`pip install opentrons`)
- Run `opentrons_simulate`

### Current behavior

[koeng@nuci3 pcr_bottleneck]$ opentrons_simulate
Traceback (most recent call last):
  File "/home/koeng/.local/bin/opentrons_simulate", line 5, in <module>
    from opentrons.simulate import main
  File "/home/koeng/.local/lib/python3.11/site-packages/opentrons/__init__.py", line 11, in <module>
    from opentrons.hardware_control import (
  File "/home/koeng/.local/lib/python3.11/site-packages/opentrons/hardware_control/__init__.py", line 14, in <module>
    from .api import API
  File "/home/koeng/.local/lib/python3.11/site-packages/opentrons/hardware_control/api.py", line 33, in <module>
    from .backends import Controller, Simulator
  File "/home/koeng/.local/lib/python3.11/site-packages/opentrons/hardware_control/backends/__init__.py", line 1, in <module>
    from .controller import Controller
  File "/home/koeng/.local/lib/python3.11/site-packages/opentrons/hardware_control/backends/controller.py", line 20, in <module>
    import aionotify  # type: ignore[import]
    ^^^^^^^^^^^^^^^^
  File "/home/koeng/.local/lib/python3.11/site-packages/aionotify/__init__.py", line 5, in <module>
    from .base import Watcher
  File "/home/koeng/.local/lib/python3.11/site-packages/aionotify/base.py", line 10, in <module>
    from . import aioutils
  File "/home/koeng/.local/lib/python3.11/site-packages/aionotify/aioutils.py", line 122, in <module>
    @asyncio.coroutine
     ^^^^^^^^^^^^^^^^^
AttributeError: module 'asyncio' has no attribute 'coroutine'. Did you mean: 'coroutines'?

### Expected behavior

Dependencies should be updated.

### Operating system

Linux

mcous, 2023-01-18:
Thanks @Koeng101, this is on our radar internally, too. For Opentrons folks, please see [RSS-172](https://opentrons.atlassian.net/browse/RSS-172)

rickwierenga, 2023-01-19:
This is an issue in the aionotify library, most likely not Opentrons (although both could be true). I opened an issue there: https://github.com/rbarrois/aionotify/issues/18

rickwierenga, 2023-01-19:
Opened a PR: https://github.com/rbarrois/aionotify/pull/19.

In the meantime, my fork (https://github.com/rickwierenga/aionotify) passes the tests. You should be able to use that to test this repo.

rickwierenga, 2023-01-28:
Hi, is there any update on this?

antoniolopezcerrato, 2023-02-23:
Hi any way around this? Need some help.

alexjoel42, 2023-02-23:
Hi! 

There is a workaround (working on this in Support as well). 

You can leverage a tool like [Pyenv](https://learnpython.com/blog/change-python-versions/) to switch Python versions and it should work! 

Sorry, this is a bit of a tricky workaround. If you go into Support's queue we'll be able to provide a bit more specific back and forth. 

[image]

caroline-wyzer, 2024-01-31:
Any updates here? Trying to automate some processes and this is a real sticking point; we need to use 3.11

Koeng101, 2024-04-09:
> Thanks @Koeng101, this is on our radar internally, too. For Opentrons folks, please see [RSS-172](https://opentrons.atlassian.net/browse/RSS-172)

Has this been on the radar for over a year?

devonbirdseye, 2024-04-24:
Is there a fix for this issue aside from downgrading to Python 3.10?

caroline-wyzer, 2024-04-24:
Doesn't seem to be. Also, to be clear (for the Opentrons folks), it's not just running opentrons_simulate that this breaks, which would be annoying but not that big of a deal--we've been trying to interface with multiple robots through one computer, and we can't import opentrons properly, so in order to get data off any robot, we have to write to text files and pass them back and forth, which takes much longer (up to several minutes instead of seconds) and is more computationally intensive and prone to errors. A long-term solution to this would be much appreciated.

devonbirdseye, 2024-04-25:
I contacted Opentrons and was told I must downgrade Python for it to work. They linked this bug ticket from January 2022, which hilariously is flagged as "closed" with the "solution" of simply downgrading python versions. I guess they don't plan on ever fixing this.

https://github.com/Opentrons/opentrons/issues/9331

alexjoel42, 2024-04-25:
If this helps, we have [HTTP API 
](https://github.com/Opentrons/opentrons-integration-tools/tree/main/http-api)

Which might fit your goals more. 

Can we talk through a bit more what you need in 3.11 so I can make a more compelling case for it?

Koeng101, 2024-04-25:
> Can we talk through a bit more what you need in 3.11

I want `opentrons_simulate` to work on the default installed version of python on the vast majority of systems. Like, yes, I *can* just run `poetry` or `pyenv`, but it complicates the systems and adds another dependency management step. The solution of "just delete python3.11 or use pyenv" ignores the fact that this is a very annoying customer-facing issue that has multiple people complaining about it.

rickwierenga, 2024-04-25:
while using the api does not strictly require any python version, it is practically necessary to import `opentrons` for the resource definitions. i understand there are ways to circumvent this, but it's not a great experience.

the fact that there is so much interest in this thread alone means it's worth solving. + you don't want to be stuck on 3.10 forever. asking users for specific motivations seems a distraction from doing the actual work, which shouldn't take that much time.

caroline-wyzer, 2024-04-25:
On our end, we've paired up robot control with many other of our internal systems and database in order to track errors, easily transfer complicated protocol information back and forth, etc, and so I would need to build pyenv into a complicated existing system. I write a lot of code, but I'm a biologist with very little formal programming training--I can definitely figure out how to make it work, but it would take me a few days to work out all the kinks. HTTP API looks really cool and useful, but is not at all integrated with our current system. Obviously I could use it, but again, this would take me several days of work, including learning the first thing about HTTP, which I currently know next to nothing about. Multiple days of work is a LOT for a bug that should be pretty easily fixed, or even just patched so opentrons can at least be imported on 3.11 without immediately breaking. And sure, if it was just me, I'd figure it out. But clearly there are a bunch of people here who are spending large chunks of their time trying to solve this issue for themselves--which is not that surprising to me, since the website documentation has a whole section designed for mid-level users recommending the use of opentrons_simulate despite needing a really annoying workaround to function. It could get fixed once now, or repeatedly cause customers relatively new to coding hours of work.

rickwierenga, 2025-07-21:
@mcous any idea when this will be fixed?

until then we are forced to use python 3.10 which is like 4 years old

closed 2025-12-15 by sfoster1