Opentrons/opentrons #12007: bug:

jeffhsu3, 2023-01-17:
### Overview

In the opentrons app stuck on "Analyzing protocol" or "".  Running the protocol on the OT-2 via opentrons_execute or opentrons_simulate succeeds however.  I tried to isolate what the issue was, it appears that dispensing 0 uliter result in the issue.  

### Steps to reproduce

```
metadata = {
    "protocolName": "Minimal Error",
    "apiLevel": "2.13",
}

def run(ctx):
    tips = "opentrons_96_filtertiprack_200ul"
    tips300 = [
        ctx.load_labware(tips, slot, "300µl tiprack")
        for slot in ["2", "3", "8", "9", "10", "11"]
    ]
    m300 = ctx.load_instrument(
        "p300_multi_gen2", 
        "left", 
        tip_racks=tips300
    )
    plate = ctx.load_labware("nest_96_wellplate_2ml_deep", "1", "plate")
    res1 = ctx.load_labware("nest_12_reservoir_15ml", "7", "reagent reservoir 1")

    all_tips = [well for rack in tips300 for well in rack.rows()[0]]

    m300.pick_up_tip(all_tips[0])
    m300.dispense(m300.current_volume, location=res1.wells()[0].top())
    m300.drop_tip(all_tips[0])

    m300.pick_up_tip()
    m300.aspirate(200, res1.wells()[0])
    m300.dispense(200, plate.wells()[0].top())
    m300.drop_tip()

```

### Current behavior

Never stops "analyzing protocol" or the stop button remains unclickable.  Not sure what causes what.  In the example above the start button is unclickable.  opentrons_simulate results in no errors.    

### Expected behavior

Start the protocol or throw error.  

### Operating system

Linux

ecormany, 2023-01-17:
Thanks for reporting this bug. [A PR to address it](https://github.com/Opentrons/opentrons/pull/11969) has been merged and it should be fixed in the next release.

mcous, 2023-01-18:
Closing as fixed. Issue will be resolved by the 6.2.1 release

closed 2023-01-18 by mcous