Opentrons/opentrons #12135: 6.2.1 tip length calibration will not save bug:

arstier, 2023-02-10:
### Overview

Tip length cannot be calibrated in app. Robot will perform calibration prior to protocol run, but will not save results.
[image]
[image]

### Steps to reproduce

Updated app to 6.2.1
Updated robot to 6.2.1
[platform was moved and pipettes were changed]
Calibrate Deck
Install new pipettes
Calibrate pipettes (including tip length)
Loaded protocol
->Alert, tip length P300 8-channel gen2 not calibrated
Calibrate tip length
Last two steps repeat
[image]

### Current behavior

Will not allow for tip calibration and does not allow for protocol operation

### Expected behavior

tip length calibration should perform and save to robot, allowing for protocol operation

### Operating system

Windows

### System and robot setup or anything else?

connected through USB connection.
Have attempted re-installation of app and robot versions which did not solve the issue.

alexjoel42, 2023-02-13:
Hi! 

Sorry that you've been experiencing this issue. 

Would you be able to email Support at support@opentrons.com with:

- The GitHub ticket you've made 
- The protocol file you're using 
- If this happened before, or if it's your first time running a protocol on the OT-2

Then we'll make sure to help you out ASAP.

arstier, 2023-02-13:
A second factor reset followed by reinstalling the 6.2.1 update onto the OT-2 platform seems to have resolved this issue.

A deck calibration was performed followed by both tip and offset calibrations which were then properly saved to the system.
[image]

The same test protocol was loaded and performed without error suggesting troubles have been resolved.

closed 2023-02-13 by arstier