Opentrons/opentrons #12166: feat: Allowing (not forcing) option to do Labware Position Check on both pipettes

RubenHitV, 2023-02-16:
### Overview

Related to bug issue #12165, posted  by me, would like to opt for the option to perform Labware Position Check on both pipettes, as calibration data somehow does not translate perfectly to protocols, causing offset issues that result in failing protocols (e.g., due to tip boxes that are tipped over to the side).

### Implementation details

Labware position check option for both pipettes for advanced protocols that use both pipettes, not just the one on the left. If this is an option, not a requirement, each user can determine if this is necessary for their purposes or not and will not cost them additional time.

### Design

Check box before Labware Position Check to click which pipettes the user wants to do the check one (either one, or both).

### Acceptance criteria

Doing the Labware Position Check and using this offset information correctly for the protocol.

danolson1, 2023-02-22:
I support this.

maartje-git, 2024-06-21:
why is this still not an option 🫤