Opentrons/opentrons #12199: bug: Building executable with PyInstaller hits "SharedDataMissingError" on execution

SPC-Archer, 2023-02-27:
### Overview

I am building a python-based GUI app which includes a feature that simulates an OT2 script.

I want to build this into a standalone app using [PyInstaller](https://pyinstaller.org/en/stable/index.html) but at runtime I hit an exception. 

I am only importing `from opentrons.simulate import simulate, format_runlog` but this also seems to import the hardware portions of the app and this eventually reaches an exception when it cannot locate the shared data.

I am not 100% sure if this is a bug or a quirk of "freezing" the Opentrons api for turning it into a standalone app so what I am wondering is:

1. Does simulation require the `SharedData` to execute?
2. If so, can I manually set `OT_SHARED_DATA_PATH` to fix this error?
3. Are there any other workarounds?

### Steps to reproduce

1. Create .py with `from opentrons.simulate import simulate, format_runlog` (only code necessary)
2. Run  `pyinstaller --name='MWE' --debug=imports  MWE.py `

The OT2 relevant portions of the traceback:
```
  File "opentrons/__init__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/hardware_control/__init__.py", line 13, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/hardware_control/adapters.py", line 6, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/hardware_control/protocols/__init__.py", line 8, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/hardware_control/protocols/liquid_handler.py", line 6, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/hardware_control/protocols/instrument_configurer.py", line 7, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/hardware_control/pipette.py", line 15, in <module>
  File "<frozen importlib._bootstrap>", line 1042, in _handle_fromlist
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "opentrons/config/pipette_config.py", line 90, in <module>
  File "opentrons_shared_data/pipette/__init__.py", line 25, in model_config
  File "opentrons_shared_data/pipette/__init__.py", line 31, in _model_config
  File "opentrons_shared_data/load.py", line 56, in load_shared_data
  File "opentrons_shared_data/load.py", line 47, in get_shared_data_root
opentrons_shared_data.load.SharedDataMissingError
```

### Current behavior

The current behaviour is that `opentrons.simulate` is not compatible with pyinstaller.

### Expected behavior

I would like `opentrons.simulate` to be compatible with pyinstaller.

### Operating system

Mac

mcous, 2023-02-27:
- "Shared data" is the `opentrons_shared_data` package. It is absolutely required for protocol analysis, as it contains various important geometric data files
    - You will probably need to configure Pyinstaller to grab these data files and add them to the bundle
    - Examining the [setup.py](https://github.com/Opentrons/opentrons/blob/e0e6cab70d444835f730c355230aee0592da7862/shared-data/python/setup.py) for `opentrons_shared_data` may make this easier to figure out
- `opentrons.simulate` is considered "legacy" code on our end; I would look at `opentrons.cli` for a more future proof method of running protocol analyses. This is the module that the Opentrons App uses

We investigated `PyInstaller` briefly for bundling the analysis system with the Opentrons App. We chose not to use it! We also investigated PyOxidizer, and ended up using a portion of it in our app.

To run protocol analyses in the Opentrons App, we bundle an [isolated Python installation](https://python-build-standalone.readthedocs.io/en/latest/) into the app and `pip install` the `opentrons` and `opentrons_shared_data` packages into that environment.

SPC-Archer, 2023-02-28:
Thanks for the quick response. 

Given `opentrons.simulate` is legacy code, I think it best that impliment a future-proof solution as opposed to putting a band-aid on my current solution.

I have had a look at what I think you mean by `opentrons.cli` but I am not sure I have grasped what it is. I have had a play with the CLI `opentrons_simulate` mentioned [on your simulation support article](https://support.opentrons.com/s/article/Simulating-OT-2-protocols-on-your-computer?). This works and I am sure that I could implement a solution using `subprocess` and piping the output. However, it looks like this is calling a python function (evidenced by the python traceback displayed when attempting to analyse a flawed protocol) so it seems silly to use python to call python via a CLI command. 

Is the right tool? I had a look around the Opentrons App source and couldn't find the relevant section that initiates the protocol analysis so I couldn't verify (but I am also not versed in React apps or JavaScript so I didn't really know where to look or what I was looking at).

mcous, 2023-02-28:
Hey @SPC-Archer, take a look [here](https://github.com/Opentrons/opentrons/blob/e0e6cab70d444835f730c355230aee0592da7862/api/src/opentrons/cli/analyze.py) for the Python source of the newer `python -m opentrons.cli analyze` pipeline.

The key calls are basically this:

```python
from pathlib import Path

from opentrons.protocol_reader import ProtocolReader
from opentrons.protocol_runner import ProtocolRunResult, create_simulating_runner

async def analyze(input_files: List[Path]) -> ProtocolRunResult:
    protocol_source = await ProtocolReader().read_saved(
        files=input_files,
        directory=None,
    )
    runner = await create_simulating_runner(robot_type=protocol_source.robot_type)
    analysis = await runner.run(protocol_source)

    return analysis
```

Big word of warning: these are internal APIs of the `opentrons` package! They aren't guaranteed to stay stable between releases. It might seem a little silly to use Python to shell out to a CLI that uses Python, but that API is likely to remain more stable than the package internals.

You can give the new CLI a try in your terminal using a Python environment with `opentrons` installed:

```shell
python -m opentrons.cli analyze --help
```

You should see something like:

```
Usage: python -m opentrons.cli analyze [OPTIONS] FILES...

  Analyze a protocol.

  You can use `opentrons analyze` to get a protocol's expected equipment and
  commands.

Options:
  --json-output PATH  Return analysis results as machine-readable JSON.
  --help              Show this message and exit.
```

For now, you _must_ use the `--json-output` option, which writes the analysis to the given JSON file.

SPC-Archer, 2023-03-01:
Thanks - this makes a lot of sense. 

In the interests of modularity, I'll use the CLI to ensure that my codebase is unlikely to be broken by future changes to your codebase. 

I have just tested it and it works nicely. 

Ideally, I would like to have a formatted runlog like is easy to produce from within `opentrons.simulate` - is it possible to easily transform the JSON output? I tried feeding it directly into the `format_runlog` function but it does not seem to be in the correct format for that. 

Not a huge deal breaker if that is not currently possible as it looks like the CLI is under active development. 

Thanks again for all your help.

SPC-Archer, 2023-03-01:
Sorry if this is a little off topic but is there a plan to allow for protocol execution via the CLI? That would be something I would be interested in. My ideal scenario would be to be able to specify protocol path, robot, and to only run if there are recent labware offsets for all labware on the deck.

mcous, 2023-03-01:
> Ideally, I would like to have a formatted runlog like is easy to produce from within opentrons.simulate - is it possible to easily transform the JSON output?

That is the longer term intention, yes. The idea (in my head) would be that if one omits the `--json-output` option, you get a human-readable output instead, similar to hoe `opentrons_simulate` currently works.

> Sorry if this is a little off topic but is there a plan to allow for protocol execution via the CLI? That would be something I would be interested in. My ideal scenario would be to be able to [...]

I think that would be a great idea! In the mean time, you _should_ be able to write your own execution CLI that uses `request` (or similar) and the robot's HTTP API. This would be my hope for our own first-party CLI, too.

You can check out [these integration docs and examples](https://github.com/Opentrons/opentrons-integration-tools/tree/main/http-api) for how to go about doing that. The basic procedure is something like:

1. `POST http://${robot_ip}:31950/protocols` with protocol file(s)
    - Get back an ID for that protocol
    - Protocol will be persisted on the robot for future use
    - This will also kick off a robot-side analysis of the protocol, available via `GET /protocols/:protocol_id/analyses`
2. `POST /runs` with protocol ID and other data, like labware offsets
    - Get back an ID for the run
    - You may create multiple runs over time from the same protocol ID
3. `POST /runs/:run_id/actions` to start, stop, or pause the run

While the run is going, you can use the following endpoints (among others) to check the status of the run

- `GET /runs/:run_id` - general run status and equipment information
- `GET /runs/:run_id/commands` - get the commands list and status of each command

The HTTP API is how the Opentrons App itself talks to the robot, and it has its own versioning scheme via the `Opentrons-Version` HTTP header, so it should remain fairly stable across software versions.

SPC-Archer, 2023-03-01:
After a bit more investigation it seems that the current generation of the CLI does not check the script for errors (it had no problem with me referencing a well that does not exist). As this is the primary reason for me to simulate the script, I am happy to wait until the development of the CLI matures before I move the codebase over.

In the mean time, I have decided to avoid packaging and use a hacky Apple Script to make an "executable" which just calls the python script. As a result, I'll close this issue as I feel my problems have been resolved. 

--

As for the prospect of using the CLI to control the robot - it sounds easy enough to create a simple interface to upload and run a script. Thanks for pointing me in that direction - this should save us a bit of time on each experiment. I'll hopefully have some time in the next few weeks to attempt that implementation.

closed 2023-03-01 by SPC-Archer